SpenceKonde/ATTinyCore #718: Attiny x61 page only has old pin layout picture

TuckerMacor, 2022-08-20:
new pin mapping seems to be missing on this page: [https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-dev%3CThis-is-the-head%2C-submit-PRs-against-this/avr/extras/ATtiny_x61.md](https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-dev%3CThis-is-the-head%2C-submit-PRs-against-this/avr/extras/ATtiny_x61.md)

SpenceKonde, 2022-08-20:
And that's an "Eeeew, I stepped in that?!" pin layout too. Not as bad as the old 167 pinout (which was not created by a human mind - we don't possess that capability for evil - that pinout could have been inspired by none other than the devil himself. In fact, the pin mapping was very nearly pessimal)

SpenceKonde, 2022-12-24:
Is anyone intererested in helping out here opr is is me and MS paint?