SpenceKonde/ATTinyCore #719: Missing Extras directory when updating from Adruino IDE

bb5317, 2022-08-22:
When updating the ATtinyCore from within the Arduino IDE the avr/extras are not copied to the appData directory which means that the user has no visibility of the pinouts and additional details documents.
Only by downloading from github can these be accessed. Is this intentional? If so, it means an Arduino IDE update does not give you all of the files necessary to use your core and the devices it supports without coming to github (which newbies may not realize).

SpenceKonde, 2022-08-22:
Arduino installs the cores what, 6 levels deep in a hidden folder that is not clearly documented and has no access link from within the IDE? They can't expect people to be viewing the files from there, so I deduce that they expect people will view the docs online. 
Since the image files are a large portion of the download (especially on the other two cores) I thought I was doing people a favor - I only set out to delete the image files (I think it's DxCore that prompted me to put this into the release script), and I coould compress those PNGs more sure, but it's lossy compression, I don't have the original images and I keep having to hack them up in Microsoft paint to add things to them. 

I wish a program like old Micrographix Designer existed, something with an easy learning curve that would be working with shapes and objects  (a square, an arrow, a line) instead of "some pixels that happen to make a box containing some text)

bb5317, 2022-08-22:
I see now that there is a "more info" link that the Arduino IDE board manager shows with the ATtinyCore that leads here to github. I'll close this issue.

closed 2022-08-22 by bb5317

SpenceKonde, 2022-08-23:
Yeah - the reason I remove extras (I didn't ask for all of extras to be deleted, just the images. All the cores use a very similar release script, ya see, which was a contribution; it just deletes  all of extras, it should probably be adapted to only get the images - but of course now there is extras/development that needs to go, and probably extras/ci - for different reasons (would you want to explain to someone one of the ci compile test sketches in dxcore to someone who created an issue alleging it to be located or used inappropriately, for example? those are *not* normal sketches). But it was delivered as all of extras being deleted and I haven't felt the situation serious enough to justify changing. The IDE hides the place where it installs packages to what, 6 levels deep inside a hidden directory or something?) But on DX core, pinout images are 44 MB of a 47MB download - that's as much as a whole toolchain update! Github scales the images automagically when displaying them normally, but retaining the massive ones helps compensate for the fact that I don't have those images in any sort of vector format nor do I even know of any such drawing tool still in active production 20 years ago the computers of the day ran software that I knew very well and would have worked a treat, but that sort of drawing tool doesn't seem to exist anymore, 

So now I am left with only paint and the originals of those images, and depend on the kindness of others for the diagrams.  I do not know what software @MCUdude used to create the pinout images for this core or what @freemovers used for megaTinyCore. I inquired as to their methods some time ago, and I do not remember the response, only that it was not actionable for me at the time. I can tell you more about how to transmute lead into gold than about how to make the pinout diagrams...

SpenceKonde, 2022-08-23:
Since the IDE clearly doesn't want people looking at files within board manager installations (otherwise it wouldn't be buried like it is), It does not seem necessary, appropriate, or helpful to include the documentation with board manager installs. Manual installations, where the fileswill be readily visible to the user in an obvious location do include them (otherwise there would like, not be any docs, because that's how I check them in)

neo305, 2023-01-17:
where do I put these files? ATTinyCore-1.5.2?