SpenceKonde/ATTinyCore #720: LED_BUILTIN missing from pinout diagrams

bb5317, 2022-08-22:
Please add the LED_BUILTIN definition to all of the pinout diagrams and maybe a note in the device readme files. It took me half of the morning trying to find the definition for a tiny828 and noticed it was missing from the diagrams. For those looking, I found it in cores/tinymodern/core_pins.h.

SpenceKonde, 2022-08-23:
This core is in a very strange state right now. 2.0.0 has been in development for close to two years and is planned for release in Q4 of 2021, so as you can see i'm a bit behind schedule.

2.0.0 is a major version change. Sketches may require modification to work with the new core (specifically analogRead() passed a number reads the digital pin number in 2.0.0, but the analog channel number in 1.x.x. 1.x.x also contains two cores when only a single core is necessary, appropriate, or justifiable, but at the time that I first got this working, I lacked the skill and knowledge to avoid that situation. 

Also, the original version made a critical mistake when selecting the optiboot baud rate. 115200 baud is not a good baud rate for 16 MHz, and this error is in the same direction as the 

2.0.0 also supports new pin mapping option for 1634, (the one I should have used in the first place, not the moronic backwards one I used) and makes the good pin mapping option the default for the 841/441. 

Those are the only intended potentially breaking changes. However, the scale of internal changes and feature enhancements is of an unprecedented scale - but being tripped up by a number of problems, and taking a backseat to the newer parts which euhm, kind of blow the doors off the classic AVRs. 32k flash 3k SRAM, 12-bit differential ADC with PGA and 6 PWM channels, plus 4 CCL logic blocks, on an AVR in a 3mm x 3mm 20 pin QFN? And that's just the tinyAVR line, they're a buck a pop for the cadillac version. Oh and they have 16/20 mhz internal oscillator, that if you set to 20 and then crank the calbyte up, at 5v the extended temp range chips sing like a bird at 32 MHz. And if that's not enough pins for your application there's th AVR Dx-series, with top end part having 55 usable I/O pins 128k flash, one port running at a voltage independent of the system voltage (MVIO, essentially builtin level shifting with perfect integration) and a trio of onchip opamps. You can even use one of them with the dac (oh yeah, the Dx has a 10 bit dac) to generate the voltage you're using for MVIO in case you're running some part that insists on a stupid voltage (pin current limits still apply of course, but they put in an opamp current chart that showed the voltage drop all the way to the abs max per opin current, not just 20mA), Spec'ed for 24 MHz operation across the full voltage range (1.8-5.5v).... but if you just increment the register that selects the oscillator speed (in units of 4) to continue the pattern, 28 and 32 mhz settings that are basically universally functional are right there, and I've got an extended temp range DB that's rock solid stable at 48 MHz from an external clock, and it's very rare to find a chip that malfunctions at 40 MHz on a crystal. And they've released the IO headers for the EA series and figured out what they're tinkering with there - but what's really whack is that the EA went back to the 16/20 MHz and voltage-dependant maximum speed, like tinies. And everyone's on the edge of their seat for product brief for the next release - I think the next thing we're gonna get after EA is a DU with USB (like they teased about - posted the product brief then pulled it down hours later), native usb. And they didn't fall for the high pincount trap with USB, the DU product brief had DD-like pincounts, 14, 20, 28 and 32 pins - the 20 pin version I think is gonna be the one that really gets people going, 3mm square, native USB, and it's an AVR. 

This isn't a place for a discussion of the relative merits of ARM vs AVR, but they each have tasks they are good at and those they are not - in computing it's almost ubiquitous as the computational ability increases, and the clock speed rises, your connection to the IO gets squishier and less solid and your ability do arbitrary I/O degrades and max I/O voltages fall and armies of level shifters must be mustered if you need to work at 5v.

Anyway, LED_BUILTIN is already listed on all of the part specific documentation pages on the 2.0.0 branch, (which as of this weekend was defined as the master branch in an attempt to get people to stop submitting PRs against the old codebase, since those PRs, If I merged them, would never be released, because they were made against the old master branch, which was 1.5.2, but alll future releases are coming from 2.0.0 branch) as it will be in 2.0.0, which is the next planned release. (Oh - this is another area of potential breakage - I recall finding a few places where the LED_BUILTIN define referenced a pin other than the pin connected to the LED in extant commercially available hardware, though though nothing even remotely common.  There will be no 1.5.3. nor was any such version ever planned. 

There are no plans to add markers to the pinout diagrams and I think where the target has a commercial board available it's already marked; there are complications relating to this for some parts as noted in the part specific documentation, I only have pinout images compressed with lossy jpg compression, and at relatively low resolution  (certainly compared to Dx-core and it's 40MB of pinout images) and let's face it, outside the context of the micronucleus boards, which are a small minority of the application of this core, we're targeting chips, not boards. There isn't a built-in LED on the chip, and who the hell knows what pin the user wired the led to. So what does LED_BUILTIN mean there? Generally, it's the "least useful pin", on classic tiny [... cut ...]
The changes in 2.0.0 are do large, I feel like I should be using gratuitous profanity to characterize the magnitude of the changes they're fucking massive. Over 500 more bootloader binaries to support a shitload of extra configurations, every part supports everything it can support, and I basically rewrote the assembly for the builtin software serial (not SoftwareSerial) about 1.5 times. the old code was awful. but now the one part that supports it (x61) can move RX to any of 3 pins, and I realized it could be extended to every part to give you the option to move it to int0 or (where present int1) in exchange for attachInterrupt() (because that's the same pin attachInterrupt can use)

bb5317, 2022-08-23:
WOW! Believe me when I say how much I appreciate your contribution to the Arduino world. It makes it feasible to use small parts for simple apps without getting into Atmel Studio and having to code everything. I also look forward to these new tiny parts but availability is the pits right now. So I have to work with the parts I have on hand until availability normalizes which looking at Digi-Key & Mouser expected arrival dates puts it about 3rd quarter 2023!  
  With the LED pin being documented in 2.0.0 I'll close this request. I was only complaining because the part (ATtiny828) I was working with didn't have it defined in the pins_arduino.h files. I eventually found it in the cores/tinymodern/core_pins.h file. 
Thank you again for your contribution.

closed 2022-08-23 by bb5317

SpenceKonde, 2022-08-24:
I'll check that it's defined in he 2.0.0 branch, I thought that was caught and fixed. because as you can see the pin_arduino files are like actually consistent in 2.0.0

SpenceKonde, 2022-08-24:
No, that's just when people who order now will get them, 

If you wait until next september, you'll see that it's backordered until Q3 of 2024. You have to be constantly checking the three big sources. Digikey, Mouser, and Microchip direct and you have to check the lead time on all versions of the part. Sometimes you'll find some september 2022 shipping version of a part from one particular vendor when all the other versions won't ship until next year. Sometimes you even find them unexpectedly in stock (microchipdirect in particular - their leadtimes are the longest - but sometimes out of the blue product that was backordered appears for immediate shipping. I think that's when someone cancels or they have a few extras because yield was unexpectedly high. All three of them have sent me an email in the past month apologizing for delayed shipping or changes to the expected ship date of my order. I don't think people get to move forward in line to take their place, I think if you buy the lot that is expected to ship on january 21, you'll get that lot, probably shipped late, but if soemone before you cancels a huge order, that won't change a thing, except that they'll suddenly be earlier slots available in line :-P

. Maybe instead of buying out so many competitors Microchip should have built more fabs.