SpenceKonde/ATTinyCore #723: An error in the http://drazzy.com/package_drazzy.com_index.json file.

yendoric, 2022-09-12:
extra comma at end of line 3243
causes:
"SyntaxError: JSON.parse: unexpected character at line 3242 column 11 of the JSON data"

kelogs, 2022-09-12:
Could this be why Arduino IDE won't show up the ATTiny boards anymore ? tried to manually (re)install it without success either

ecrzgz, 2022-09-12:
We can´t see ATTiny boards in the arduino ide, anybody know how do we resolve it?

theosteini, 2022-09-12:
Yes because of this you can't see the ATTINY boards. 
To solve this go to Documents/ArduinoData. Open the .json file: package_drazzy.com_index.json 
Go to line 4241 and remove the comma and save the file. After that everything should work again. Maybe you have to open the library manager in the tools section of your Arduino IDE so that the librarys update.  
[image]

kelogs, 2022-09-12:
@theosteini yes, the libraries do update, causing a fresh copy of the .json to be downloaded. Any way around it ?

Eventually, after moving boards.txt to the variants subfolder, Arduino IDE started showing the ATtiny boards but would not compile anything:

`loading hardware from /home/kellogs/Sensory/hardware: loading package avr: invalid version dir /home/kellogs/Sensory/hardware/avr/extras/development: no major version found`

ecrzgz, 2022-09-13:
@theosteini, Thank you very much. I think so everythink wprk properly now

yendoric, 2022-09-13:
Do as theostini suggests, but he has line number wrong.
Save file and then make read-only.
Then try to remember to delete it when the file is eventually fixed.

SpenceKonde, 2022-09-17:
issue with the was quiickly fixed

closed 2022-09-17 by SpenceKonde