SpenceKonde/ATTinyCore #735: SPI errors when compiling Adafruit ADS1x15 I2C library for ATTiny85

nodemand, 2022-12-20:
When trying to compile or verify the Adafruit ADS1x15 library for ATTiny85 (no bootloader) the Arduino IDE (2.0.3) returns the following SPI errors while it does not even use SPI. It does compile on ATTinyCore 1.5.2.

`
Using board 'attinyx5' from platform in folder: /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr
Using core 'tiny' from platform in folder: /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr
Detecting libraries used...
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-237199723B74F710A349D90E42CA7047/sketch/singleended.ino.cpp -o /dev/null
Alternatives for Adafruit_ADS1X15.h: [Adafruit ADS1X15@2.4.0]
ResolveLibrary(Adafruit_ADS1X15.h)
  -> candidates: [Adafruit ADS1X15@2.4.0]
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-237199723B74F710A349D90E42CA7047/sketch/singleended.ino.cpp -o /dev/null
Alternatives for Adafruit_I2CDevice.h: [Adafruit BusIO@1.14.1]
ResolveLibrary(Adafruit_I2CDevice.h)
  -> candidates: [Adafruit BusIO@1.14.1]
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-237199723B74F710A349D90E42CA7047/sketch/singleended.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire@2.0.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@2.0.0]
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/Wire/src /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-237199723B74F710A349D90E42CA7047/sketch/singleended.ino.cpp -o /dev/null
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/Wire/src /Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15/Adafruit_ADS1X15.cpp -o /dev/null
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/Wire/src /Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO/Adafruit_BusIO_Register.cpp -o /dev/null
Alternatives for SPI.h: [SPI@2.0.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@2.0.0]
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/Wire/src -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/SPI /Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO/Adafruit_BusIO_Register.cpp -o /dev/null
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=20000000UL -DCLOCK_SOURCE=1 -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 -I/Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO -I/Users/jeroen/Dropb [... cut ...]
Using library Adafruit ADS1X15 at version 2.4.0 in folder: /Users/jeroen/Dropbox/Arduino/libraries/Adafruit_ADS1X15 
Using library Adafruit BusIO at version 1.14.1 in folder: /Users/jeroen/Dropbox/Arduino/libraries/Adafruit_BusIO 
Using library Wire at version 2.0.0 in folder: /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/Wire 
Using library SPI at version 2.0.0 in folder: /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/libraries/SPI 
exit status 1

Compilation error: exit status 1
`

SpenceKonde, 2022-12-21:
Ugh, somehow the pins aren't propagating to the libraries correctly...... Probably not a hard fix

designer2k2, 2022-12-27:
Hello,
to confirm it, the example "BarometricPressureSensor" is giving the same error as mentioned above.  

CI is now showing it: https://github.com/SpenceKonde/ATTinyCore/actions/runs/3785433780/jobs/6435473082

SpenceKonde, 2022-12-30:
Yeah I will be committing a fix soon

SpenceKonde, 2023-06-20:
This is belived to be sorted now.

closed 2023-06-20 by SpenceKonde