SpenceKonde/ATTinyCore #738: I2C Lidar Sensor module (VL6180X) does’t work reliably in Attiny85 at 1Mhz

Rimbaldo, 2023-01-07:
I’m doing a project using a bare Attiny85 and an I2C Time of flight sensor (VL6180x).
 
The module has 10k pullups on the SDA and SCL lines, but I also added 10k resistors externally on the SDA and SCL pins.

I put a 0.1uF ceramic cap between VCC and GND of the Attiny85, running at 5V.

When I use a Digispark at 16.5Mhz, the sensor works fine. When I use the bare Attiny at 8Mhz internal, it also works normally.

But when I use the bare Attiny at 1Mhz, the TOF sensor locks a couple of seconds after running the sketch. The Attiny keeps working normally, but the sensor stops.. 

Is this “bug” related to the I2C implementation on the Attiny85? Is there any I2C configuration that I could try to make it work at 1Mhz?

hmeijdam, 2023-01-07:
I suspect that 1MHz is too slow for the AT85 to keep up with the I2C timing requirements. Keep in mind it has no hardware I2C implementation, but needs to everything in software alongside what you ask it to do else.
You are not using Tinywire or any other external library for I2C? Just the core's Wire.h

Rimbaldo, 2023-01-07:
@hmeijdam I’m using just the core Wire.h…. Should I try another I2C external library?  

I read in Hackaday below that someone changed the I2C stretch clock from 230uS to 1500us and made it work, but when using an esp266 as an I2C master and the Attiny85 at 1Mhz as a slave. Would
it work for me? How could I change this stretch clock time to try this out?

[](https://hackaday.io/project/27316-synchroclock/log/67571-attiny85-i2c-slave-at-1mhz)

hmeijdam, 2023-01-07:
I have no experience to comment on that. Why is the 1 MHz clock speed of importance to you? 

Btw, the github version of Attinycore supports more clock speeds between 1 and 8. Maybe worth experimenting with that.

[image]

SpenceKonde, 2023-01-07:
I would say no, you should not use an external I2C library. I mean hell, our Wire.h is one of the popular USI-as-I2C libraries (I don't remember off hand which one) - the main difference is that unlike the others, which do not implement a compatible API and hence are not code compatible (the differences were generally really silly too, not like "they left stuff out because supporting it was hard" but rather "they just chose to implement a somewhat different API"). The point of the Unified Wire was to gather the three different I2C implementations that were needed to support I2C on all ATTinyCore parts (a couple have a real TWI, most have a USI, two have neither a USI nor a complete hardware TWI (the 841 and 441 have a slave only hardware TWI), and a third, the 828 has a USI but can't use it for I2C because of a silicon bug - so those three parts have to use a fully software implementation. All of those libraries had non-identical APIs, particularly the software one), and put them all into one library, call it Wire.h, and make sure that the name of the object they provided was "Wire", and that the name of the class was always "TwoWire" like the official library: many libraries expect you to pass it a pointer or reference to an instance of a class named TwoWire, and expects that whatever you pass it will present the normal API. Other libraries don't even bother with that - they include Wire.h, and expect it to provide something named Wire with the normal API. 

Our Wire.h uses the preprocessor and knowledge of which AVRs support what to select one of three implementations of I2C *and is meant to be a drop-in replacement for wire* and maximize code compatibility with other code written for other AVR devices. 

But i'm going to echo the sentiment that 1 MHz may just be too bloody slow for some I2C devices. The clock timing is handled in software, and it ends up timing the high and low of SCL by a compile-time calculated number of iterations of the standard three clock timing loop, but at such low clock speeds, you start to see integer-math truncation and rounding effects. One side of the clock ends up with 2 passes through the loop (trying for as 5 us delay, getting 6us) the other tries for 4us, but gets 1 pass through the loop, 3us... Now account for the fact that each clock cycle between these is 1us, not 1/8th or 1/16th of a us, if you look at sending a byte, the SCL high positive edge happens 6 clocks after entering the USI data loop, and is done on the 7th clock, then 2 or 5 clocks in the while loop to see if the pin has gone high, then 3us with the second delay. then after 12 or 15 clocks we get to the end of the high, set by the 13th or 16th clock, which has lasted either 6 or 9 us. Now the low begins, the test is conducted, and we're back at the start. So the entire cycle is at best 16 or 19us of which 10us is low.... so you;re operating well below standard mode speeds. but it's not obviously pathological that I would expect most devices to object. The high is at least regulation length, the low is twice regulation length. It's running at 10/16ths or 10/19ths of the clock speed it's supposed to be, so it's wrong, but only a little wrong, so what? Most I2C devices will run even when SCL is much slower than that, provided the highs and lows are long enough. But occasionally one comes across particularly troublesome "I2C" devices that have problems at certain SCL speeds. The Melexis I2C IR sensors for example are notorious for that (in fact, trying to make those things work resulted in two people independently rewriting the clock configuration code in the mTC/DxC Wire, one of whom went on to reimplement most of the entire library, greatly reducing it's flash footprint and forming a more coherent basis for a the handful of methods that extend the Wire.h API. 

A scope trace of the SDA and SCL lines while it works and once it has started failing would be needed in order for me to give any further advice; the source code would also be helpful to rule out other possible problems. 

I'm not sure what is meant by changing the clock stretching. Clock stretching is something that the slave device may do: After receiving the 8 bytes of data, the slave device is permitted to hold the clock line low while it processes that byte, before deciding to reply with an ACK condition or a NAK condition. The Arduino Wire library, like most Arduino libraries was not, shall we say, designed by embedded design experts with a close eye on performance (afaict, most arduino libraries, particularly official ones, were neither designed by experts, nor with particular concern for performance). Maybe they meant lengthening the time the ESP would wait while the clock was stretched? That would make sense when trying to talk to an ATTiny85 at 1 MHz acting as an I2C slave. The AVR architecture, the limitations of C/C++, avr-gcc's questionable compiling decisions (questionable in terms of performance, not correctness), the Arduino paradigm and the API dictated by that, and many of the more boneheaded idioms common in the Arduino community - they all contribute to most cases of surprisingly poor performance, but all of these issues are at a maximum here. First - There are two interrupts associated with I2C as slave, and the way this maps to the ArduinoAPI are:
[... cut ...] s an optional complementary output pin with a dead time generator. Both of these parts are designed with motor control applications in mind, where you often need a higher pwm frequency, and you need programmable deadtime to prevent shoot-through that would burn out the H-bridge. The x5 works for 2-pole BLDC motors, while the x61 can do 2-pole or 3-pole ones The wacky timer is great if you want to drive a BLDC motor - everyone else would probably be much happier had it been a normal 16-bit timer.

SpenceKonde, 2023-01-08:
I will note that  2.0.0 will have 2 MHz and 4 MHz via internal oscillator available as a clock speed option on all parts (starts up with clock prescale set to 8 by fuse in case voltage is too low for 8, then shifts prescaler to /4 or /2 as appropriate), providing intermediate options between 1 and 8. 

Also, if you're running at 1 MHz for power consumption, be aware that you save much more power when in sleep mode than when active, even at 1 MHz. 1 MHz is still >1mA, power down sleep is under 1 uA. And the chip draws more current at 2 MHz - but not twice as much. And with the timing you end up with at 1 MHz, you're already prolonging I2C communication cycle significantly. Assuming you spend most of your time in power down sleep, and are just periodically waking up to take those measurements, if you spend 1% of your time awake at 1 MHz the average power consumption is higher than if you spend 0.5% of the time awake at 2 MHz (whether or not you account for the power consumption during power down sleep, or simply approximate it as zero. With wake cycles of around 1% zero is a good approximation of the contribution of the current in power down sleep mode assuming it is being used correctly and is reaching the expected current. Another order of magnitude, though and you start noticing it, and roughly 2 orders of magnitude lower wake fraction (ie, 0.01% awake) is around where t<sub>wake</sub>*I<sub>wake is of the same magnitude as t<sub>sleep</sub>*I<sub>sleep</sub>, marking the boundary between the regime where power consumption is dominated be the power used during the brief waking periods, and the regime where the power is dominated by the consumption while sleeping.

Remember that when you are sleeping, the clock is stopped so it doesn't matter what the nominal system clock is as far as the power consumed during sleep mode (what matters there is that you remembered to disable the ADC and that you don't have any floating pins, and that you don't have any loads connected to any outputs (obviously), and whether you have the WDT enabled (for waking up on a time interval) or BOD enabled (to hold the part in reset if the supply voltage is below the specified threshold. Note that while in BOD, the power consumption is markedly higher than power down sleep. That is to say, no, you cannot use BOD alone as UVLO to prevent damage to lithium batteries from overdischarge (BOD reset state draws orders of magnitude more current than power down sleep. 

It's objective is to prevent the processor from misbehaving, as AVRs are wont to do when their safe opperating conditetions are exceeded sufficiently egregiously (in terms of dF/VDD, VDD and .  (for example, calculating that 2+2 = 0 (non-zero results of mathematical operators  coming out as zero, and less often, mathematical results coming out with bits flipped seemingly at random. This directly (incorrect status bits resulting from an instruction,  or indirectly (bad values used in calculation) corrupts the SREG. The corruption of SREG, or simply incorrect execution of flow control instructions thus deranges program flow. Often the final moments involve execution bounding across the flash, with each call encountered stacking a possibly corrupt pair of values onto the stack and each return jumps to whatever the top of the stack contains a pointer to, which may have been corrupted before it was calculated, or when it is read, and this further deranges program flow.; usually the madness quickly reaches a point when the value at the top of the stack is 0x0000 (remember how non-zero results coming out as 0 is the most common failure mode), and execution hits a return statement, at which point execution ends up at 0x0000, and no reset cause has been set because no reset occurred. At this point, as long as the system manages to successfully evaluate the less than a dozen instructions in the path from 0x0000 to when we fire software reset because re realize the reset was dirty, it will reset cleanly. Assuming the overclocking was generated internally, this should bring part back in spec, until the sketch tries to make it perform beyond abilities again. BOR sets a voltage tjreshold if the voltage may suffer brownouts to keep the chip idle, It is unconcerned with the battery, which should have it's own UVLO to prevent overdischarge!! 

when you're in power down sleep mode, there is no system clock being generated. Power used while in power down sleep doesn't change regardless of the system clock configuration is the same whether the clock is set to 16 MHz PLL clock or 128kHz WDT clock,

Rimbaldo, 2023-01-09:
Thanks for all your deep explanations!!

I tested in 4Mhz and the I2C sensor worked normally!

I'll scope the I2C lines later at 1Mhz to see what happens and I''ll get back later to you with the data... Exactly how and what should I measure in  the lines?

Indeed my goal is to save current, that's why I wanted to use 1Mhz. I didn't think in using power saving modes (sleep modes) because I have this Attiny with the Lidar sensor connected through a 1 meter cable to another Attiny167 at the lipo Battery side. And they both exchange data through a 1 wire transfer half-duplex protocol. So they have always to be in sync with each other.

But the sensor draws more current than the Attiny... Currently at 4Mhz I'm drawing 5mA at the Attiny  and more 5-10mA at the sensor...

SpenceKonde, 2023-01-20:
They should do the same thing at 1 MHz that they do at 4. Except that they will not do it quite with the same timing; it may be obvious what it's doing wrong (i, if a high or low is coming out way short), or it may be harder to see what it;s doing wrong. but I predict that you'll see when you're doing something it doesn't like, either it won't respond at all, or it will respond by holding one of the lines (likely the clock line) low indefinitely. That's what a lot of things do when you do something on the bus that confuses them

SpenceKonde, 2023-05-01:
2.0.0 will also have a 2 MHz option, which you may find works.

if you don't care about millis, obviously, you just manually set the prescaler to 2 when talking to the lidar and back down to 8 when not doing so. 

Do note, for all AVRs, if you are CPU-bound for the one task you do while away, ie, running at speed X takes 10 seconds, and running at speed 2X takes 5 seconds - and you spend the rest of the time in powerdown sleep, you are always better running at speed 2X, because Idd(X) < Idd(2X), pretty much over the whole AVR product line. I know of no exceptions. Running at half the speed takes more than half the current (this is how it has been on every AVR I've looked into power consumption for, largely to smack down people who kept asking me for lower clock speeds to save power, and then you look at their code and they're spending all their time in power down sleep when the clock isn't even running and the clock speed doesn't matter at all and so what they were asking for was a mode that would have problems for certain due to the low speed, and would actually use MORE POWER than what they were doing now! \o/

SpenceKonde, 2023-05-01:
`I'll scope the I2C lines later at 1Mhz to see what happens and I''ll get back later to you with the data... Exactly how and what should I measure in the lines?`

If you've got a digital scope, the most useful thing would be a single trigger showing the successful transfer (with the horizontal scale adjusted so it just fits on the screen), and then at 1 MHz, you should see that something isn't coming back right. My ideal thing to compare would the the 4 MHz working one and the 1 MHz non-working one with the scope's scale adjusted so that a byte took up about the same amount of space on the screen for the two.

Rimbaldo, 2023-09-20:
Hi! Just an update... After updating to the latest 2.0 core version of three weeks ago, my VL6180X I2C Lidar sensor module is working normally in my Attiny85 at 1Mhz! In the previous core I had installed, if I got from 8Mhz to 4 and to 1Mhz, the speed (even time in seconds from millis() variables) would get slower and slower to the point of the sensor being unusable. Now at 1Mhz it works normally as in 8Mhz, with the timings also correct.

One weird thing I note, and I don't know if it's an Arduino IDE 2.2.1 or an Attiny 2.00 Core bug, is:

When I re-save the sketch I'm working with another name (a backup, for instace), most times I have to re-burn the bootloader at the desired speed, and then re-upload the sketch, even if the bootloader was burned at the desired speed beforehand. but after that it works normally. It seems that when I save some sketch with another name, it resets the speed internally to the default (8Mhz in both my Attiny's 85 and 167)