SpenceKonde/ATTinyCore #739: Compilation errors when compiling for ATTiny85 using I2C

slawrie25, 2023-01-07:
Trying to use a standalone ATTiny85 with the Adafruit MMA8451 accelerometer, example code compiles just fine on version 1.5.2, but not 2.0.0

SpenceKonde, 2023-01-08:
Thank you for the report. In order to assist the investigation, can you please:
file-> preferences. Under Show Verbose Output During, check both boxes, and under warnings, select "all" (I don't know OTOH whether I've yet rigged the platform.txt over here to always give you warnings even if you ask for no warnings (the reason is that there are far too many things in C that ought to be compile errors, and which 99 times out of 10 indicate that there is a bug there (possibly a latent bug but a bug nonetheless. Since my cores are not supposed to generate warnings (only recently figured out how to make them pragma messages instead of warnings for things like the "hey watch the pinout" warning on parts with multiple pinmapping options. Actually, no, I was told how to do that, I didn't even know it was a think until they told me about it,)

Sorry, got a bit sidetracked
Then, reproduce the problem. Copy and paste the full text out of the console and post it here, that will greatly expedite debugging. 

And the source code that is not compiling (a minimal example ideally) 

Also, a library cannot be uniquely identified by the file name. I've run into several examples of libraries with the same name, that essentially do the same thing, but present a totally different API and are totally different libraries. I even maintain such libraries! There is a Flash.h in megaTinyCore and a Flash.h in DxCore. They both write to flash from the app. They are both named Flash.h. There are no other similarities (the fact that the NVMCTRL is different is only a small part of the reason - we actually were both writing our versions for different architectures (modern AVRs with paged writes and modern AVRs with word writes) at about the same time, and had no knowledge of the others work until each of us were done with our own. 

Unless you can reproduce the bug without that library, *I need to know not just the name of the library, but where it came from*. The way github lets anyone fork anything leads to a sort of version hell:
[image]
(do you have any idea how many different, incompatible libraries exist named LiquidCrystal or LiquidCrystal_I2C which are used in projects and written up in blogs across the internet? I would estimate that it is a 3 digit number (in base 10). Worse still, notice that in that diagram, I didn't mention either B or C documenting their changes. It is not unusual to find two github repos, one forked from the other where code files have been changed significantly. The readme is untouched, without the name of the individual who's doing the new work, no explanation of the changes, properties.txt is unmodfied - yet they changed the code dramatically, the libraries are no longer compatible, and the only way to discover that it to try to use it. and then realize that there are a bunch of versions of the library and you don't know which one the original author used to make the code you're using as a reference for how to do something. I recommend that any library you downloaded, when you include it, you should put a comment identifying where the fuck it came from, typically a github URL or the library manager library name and version number. 

I got burned by this a few years back. When I suffered a harddrive crash - I had the sketches backed up, but not the libraries. I did not remember which version of the liquid crystal library I used, all I could tell is the file name was the same, and the API totally different, and I wound up having to rewrite the whole UI section of the program, because all the LCD code had to be totally redone because I couldn't find the library I used, which I had incorrectly thought was the most-maintained version (it obviously was far from it).

slawrie25, 2023-01-08:
The library is the "Adafruit MMA8451 Library" v1.2.1 installed from library manger. Sketch is the demo one included in the library.
[MMA8451demo.zip](https://github.com/SpenceKonde/ATTinyCore/files/10369458/MMA8451demo.zip)
Running Arduino version 1.8.12.
Followng errors are produced when attempting to compile for ATTiny85:

`Arduino: 1.8.12 (Windows 10), Board: "ATtiny85/45/25 (No Bootloader), -Os (size, recommended), ATtiny85, 8 MHz (internal), EEPROM retained, RX on PB1, TX on PB (default PB0), B.O.D. Disabled (saves power), Enabled"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\sglaw\AppData\Local\Arduino15\packages -hardware C:\Users\sglaw\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\sglaw\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\sglaw\Documents\Arduino\libraries -fqbn=ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this:avr:attinyx5:optimization=size,chip=85,clock=internal_8m,eesave=aenable,softserial=enable,bod=disable,millis=enabled -ide-version=10812 -build-path C:\Users\sglaw\AppData\Local\Temp\arduino_build_451349 -warnings=all -build-cache C:\Users\sglaw\AppData\Local\Temp\arduino_cache_496344 -prefs=build.warn_data_percentage=75 -verbose C:\Users\sglaw\Documents\Arduino\libraries\Adafruit_MMA8451_Library\examples\MMA8451demo\MMA8451demo.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\sglaw\AppData\Local\Arduino15\packages -hardware C:\Users\sglaw\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\sglaw\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\sglaw\Documents\Arduino\libraries -fqbn=ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this:avr:attinyx5:optimization=size,chip=85,clock=internal_8m,eesave=aenable,softserial=enable,bod=disable,millis=enabled -ide-version=10812 -build-path C:\Users\sglaw\AppData\Local\Temp\arduino_build_451349 -warnings=all -build-cache C:\Users\sglaw\AppData\Local\Temp\arduino_cache_496344 -prefs=build.warn_data_percentage=75 -verbose C:\Users\sglaw\Documents\Arduino\libraries\Adafruit_MMA8451_Library\examples\MMA8451demo\MMA8451demo.ino
Using board 'attinyx5' from platform in folder: C:\Users\sglaw\Documents\Arduino\hardware\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\avr
Using core 'tiny' from platform in folder: C:\Users\sglaw\Documents\Arduino\hardware\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\avr
Detecting libraries used...
"C:\\Users\\sglaw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DARDUINO=10812 -DARDUINO=10812 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IC:\\Users\\sglaw\\Documents\\Arduino\\hardware\\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\\avr\\cores\\tiny" "-IC:\\Users\\sglaw\\Documents\\Arduino\\hardware\\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\\avr\\variants\\tinyx5" "C:\\Users\\sglaw\\AppData\\Local\\Temp\\arduino_build_451349\\sketch\\MMA8451demo.ino.cpp" -o nul
Alternatives for Wire.h: [Wire@2.0.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@2.0.0]
"C:\\Users\\sglaw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DARDUINO=10812 -DARDUINO=10812 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IC:\\Users\\sglaw\\Documents\\Arduino\\hardware\\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\\avr\\cores\\tiny" "-IC:\\Users\\sglaw\\Documents\\Arduino\\hardware\\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\\avr\\variants\\tinyx5" "-IC:\\Users\\sglaw\\Documents\\Arduino\\hardware\\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\\avr\\libraries\\Wire\\src" "C:\\Users\\sglaw\\AppData\\Local\\Temp\\arduino_build_451349\\sketch\\MMA8451demo.ino.cpp" -o nul
Alternatives for Adafruit_MMA8451.h: [Adafruit_MMA8451_Library@1.2.1]
ResolveLibrary(Adafruit_MMA8451.h)
  -> candidates: [Adafruit_MMA8451_Library@1.2.1]
"C:\\Users\\sglaw\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK [... cut ...]
Using library Adafruit_MMA8451_Library at version 1.2.1 in folder: C:\Users\sglaw\Documents\Arduino\libraries\Adafruit_MMA8451_Library 
Using library Adafruit_BusIO at version 1.14.1 in folder: C:\Users\sglaw\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Adafruit_Unified_Sensor at version 1.1.7 in folder: C:\Users\sglaw\Documents\Arduino\libraries\Adafruit_Unified_Sensor 
Using library SPI at version 2.0.0 in folder: C:\Users\sglaw\Documents\Arduino\hardware\ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this\avr\libraries\SPI 
exit status 1
Error compiling for board ATtiny85/45/25 (No Bootloader).`

SpenceKonde, 2023-06-14:
Should be sorted now. There were still garbage references to names given to variables in the header files (which Atmel was terriubble about keeping consistent

commit d0722e58 (2023-06-14): Correct some issues in SPI.cpp

Should fix #739
closed 2023-06-14 by SpenceKonde