SpenceKonde/ATTinyCore #740: ATtiny104 Xplained Nano

aladdn16, 2023-01-09:
ATtiny104 Xplained Nano
I have this card, can I program it with attiny core?
How should I configure this card?

Which mcu supports attiny? is there a list of all supported mcus?

SpenceKonde, 2023-01-09:
See the readme, there is a section listing supported parts and unsupported
parts and linking to cores for all parts for which im aware of a core. ATTC
does not support the AVRrc parts because the flash is miniscule and fitting
the arduino api functions into it is not really possible - not without
compromises. An attempt was made, though there are severe constraints on it
because everything needs to be so aggressively optimized for flash. It is
linked from the unsupported parts list in the readme

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2023-01-13:
### Non-supported parts
ATTinyCore supports classic ATtiny parts. It does not support any other AVR devices - There is a different core for the ones that are practical for use with Arduino, depending on the product line:
* [tinyAVR 0/1/2-series](https://github.com/SpenceKonde/megaTinyCore/) Modern tinyAVR (with 0, 1, or 2 as next-to-last digit) are supported by my megaTinyCore instead. They are totally different in every way except the "t-word" in the name, and many of them have a peripheral selection to make most classic AVRs turn gteen with envy (like a 3227 or 3226's peripera;s beat the stuffing out of many classic AVR parts, including the 328pb
* [ATtiny13/13A](https://github.com/MCUdude/MicroCore/) are supported by MicroCore by @MCUdude - 2k of flash is the lower cutoff for ATTinyCore.
* The ATtiny28L is not supported by any Arduino core. Ancient, weird, and most importantly, they don't have RAM, just registers. Yeah, seriously.
* ATtiny 4/5/10/11 and any other "AVRrc" (reduced core) parts, these combine minuscule memory sizes with a gimped CPU. [Try this core](https://github.com/technoblogy/attiny10core)
* Anything with "ATmega" in the name - you want [one of MCUDude's cores](https://github.com/MCUdude/)
* AVR Dx-series (AVR128DA64, etc) - [the crown jewel of the AVR product line](https://github.com/SpenceKonde/), supported by my DxCore.

SpenceKonde, 2023-01-21:
Closing as there is already a list of supported and a list of unsupported parts right in the readme file, this part is not on it and never will be. I have no desire to make a core for AVRrc architecture.

closed 2023-01-21 by SpenceKonde