SpenceKonde/ATTinyCore #741: PWM ATTiny45 PB0 and PB1

ltwin8, 2023-01-12:
Hello,

i would like to achieve a PWM frequency between about 16kHz and 25kHz to drive two led drivers (for a CCT LED Lamp)

I would like to avoid getting audible frequency from dimming. is it possible that by default PWM on those Pins have different Frequencies?

could someone provide me with the necessery register-manipultation to acieve on both pins the same PWM Frequency and in the range between 16 kHz to 25 kHz, i would need to use internal oscillator

SpenceKonde, 2023-01-13:
The stock configuration for the PWM of timer1 has the PWM channel that can support either Timer0 or Timer1 PWM set to timer2. 

However, in accordance with the stated design principles we also try to get the PWM frequency to come in around 1 kHz, within a factor of 2, with default configuration. Why? Because if you're drivinga power mosfet with the PWM (a common use case), the fastest you can go with a high current MOSFET is around 2-8 kHz (the lower the current the mosfet can carry, the higher the frequency, because the gate capacitance tends to be proportional to current handling capacity. Above that speed, the switching losses rapidly increase, increasing waste heat at the MOSFET, and resulting in duty cycle distortion. Extreme cases will generate enough heat to burn out the MOSFET, when it wouldn't even get warm without the PWM. See https://github.com/SpenceKonde/ProductInfo/blob/master/MOSFETs/Guide.md for a much more detailed description of how and why power MOSFET performance degrades so quickly when using relatively weak microcontroller pins to drive the gates as is the common practice (the summary is - mosfet gate has parasitic capacitance to drain and source. that means that the voltage on the gate cannot change instantly but rather takes a period of time determined by the current that can be supplied to the gate while switching. Microcontroller pins are not going to give you high currents into and out of the pin, hence the fet spends longer in the region between fully on and fully off. If the fet is essentuially on at 2.0V, but you're driving it all the way to 5v, that also means duty cycle distortion, as there's a certain amount of charge tht must be removed (graphs are given in the datasheets) before it start really turning off will result in the FET taking longer to turn off (or even start turning off) which are different from the time it takes to turn on. Beyond that, turing the time between essentially fully on, and fully off, the fet is conducting a moderate to high curent, but with higher voltage drop and resistance than usual, hence generating more heat in the FET (not uncommon for me to hear from someone who burned out FETs like that). The solution is using a "gate driver" IC, which is capablr of brief but very high current pulses which are able to swtich the gate far faster, (2-6 or more AMPS is unremarkable) for the tiny snippet of time it takes to turn the FET on or off, switching time is greatly reduced, efficiency increases, waste heat generation at the fets is reduced, etc. 

The concept of switching losses in MOSFETs is something that based on very common questions, most users are really not familiar with, and a sickening number of tutorials make false assertions or neglect the topic entirely. Worse still, many of them don't even understand the permitted operating conditions well enough, such that it's common to see recommendations of TO-220 power MOSFETs that can be driven at 3.3v. Those lists are invariably wrong, unless you are willing to accept exceptionally poor performance. Until a couple of years ago there was a single option in TO-220 that WAS usable at 3.3v gate drive, (and it offered 30V  Vds!). Sadly, the IRF3708 has been discontinued, There are no non-surface mount power mosfets (as in large max current) which can be turned on with 3.3V. (as it happens, I bought a shitload of them. especially in TO-220, but also the SMD version, because it also happened to be the only one with Max Vds = 30V in an easy-to solder package but was in spec at Vgs = 3.3 with decent Rds(on). If you really need the 3.3v gate voltage, and teh 30Vds, well, things get nastier. Unless you were to make a deal with someone who happened to have excess of the good chips on hand, like yours truly (if you managed to score 3224 in TSSOP or 3226 in QFN - lets talk - be sure also to let me know if they're E-spec or I-spec, I value E-spec higher because they overclock better ;-) ). I could be persuaded to part with the TO-220 ones, for the right price, whether in the 2-series parts I missed, or in the form of holy scrolls dedicated to Mammon, the god of materialistic greed (some unbelivers call them dollars, or for emphasis, "the almighty dollar"). The unique part of that is 3.3v gate voltage, high current, eith Vds=30. If 20V was all it had to hold off, the game changed completely, as the MCU90N02, which are fully on at 1.8v, even lower resistance at 2.5 and 4.5V on gate, and withstand 20 Vds. as do several; competing products, (Though MicroCommercial, when considering all factors, comes out looking very good indeed - although their methods involve brute force afaict, and hence high gate capacitance, making them ill suited to fast switching without a gate driver.) 

So you want 8 bit accuracy (256 timer ticks per cycle) amd 25 kHZthat means 25000 x 256 = 100000 x 64 = 6.4 MHZ timer base clock. I'd instead go got 8 MHZ timer base clock, giving me 8,000,000 ticks per second divided by 256 =  31.250 kHz PWM. Gate driver would 100% be neded. Where could you get some nice MOSFETs with gate drivers? Guess what I happen to have for sale? 
https://www.tindie.com/products/drazzy/4-channel-mosfet-board-with-optional-driver/ Exactly whhat tou need for high frequwncy PWM ;-)

ltwin8, 2023-01-13:
hello, thank you for the very detailed answer, i need to drive a highZ input (few pF) so no problem with drivestreangh. could you maybe give me advise on the registers i need to change? i would like to use the same timer for B0 and B1 (i suppose wtat would be easier?)

SpenceKonde, 2023-08-13:
The latest 2.0.0-dev code - most notably the tiny85 part specific documentation's PWM section and the link to the doc on changing timer frequency while still keeping analogWrite viable.
You can actually get 8-bit PWM at a frequency that is within a factor of 2 of any arbitrary target frequency from 1.9 Hz to 250 kHz from the two timer1 pins.... (assuming 8 MHz system clock - Max prescale on that gives 1.9, no prescale on 64 MHz from the pll gives the high figure.

closed 2023-08-13 by SpenceKonde