SpenceKonde/ATTinyCore #742: Support for Arduino IDE 2

igo, 2023-01-14:
Hi, is there any plan for supporting Arduino IDE 2? Trying on M1 Mac just to compile blink example for attiny85 and got error:

```
Error resolving FQBN: getting build properties for board ATTinyCore:avr:attinyx5opti: invalid option 'LTO'
Compilation error: Error resolving FQBN: getting build properties for board ATTinyCore:avr:attinyx5opti: invalid option 'LTO'
```

After disabling LTO in menu I got:

```
Error resolving FQBN: getting build properties for board ATTinyCore:avr:attinyx5opti: invalid value '8internal' for option 'clock'
Compilation error: Error resolving FQBN: getting build properties for board ATTinyCore:avr:attinyx5opti: invalid value '8internal' for option 'clock'
```

After trying to chnging frequesncy to 1Mhz and disabling LOT I got:

```
Error resolving FQBN: getting build properties for board ATTinyCore:avr:attinyx5opti: invalid option 'TimerClockSource'
Compilation error: Error resolving FQBN: getting build properties for board ATTinyCore:avr:attinyx5opti: invalid option 'TimerClockSource'
```

SpenceKonde, 2023-01-14:
Not in the near future. 2.0 ide has many problems right now and first we
need to get version 2.0.0 of the core out in general. Hopefully by then the
bugs in ide 2.0 will have been fixed.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

hmeijdam, 2023-01-14:
I suspect I will be on IDE 1.x for a loooooong time.. 2.0 IDE does not even have a portable function, so having multiple IDE versions/variants ( I have 14 of them) is not possible.

closed 2023-01-21 by SpenceKonde