SpenceKonde/ATTinyCore #743: Lack of memory or is it.

eddy-99, 2023-01-17:
Not sure if this is the place to post but I am running out of options.
ATTiny88 dev board. Arduino 2.03
Sketch shown below will run 3-4 movements but if I add another move it seems to go into a loop.
I have tried variations of the code, the sketch uses between 40-60% of memory.
Is it just running out of memory on the ATTiny?

Sketch uses 4258 bytes (62%) of program storage space. Maximum is 6780 bytes.
Global variables use 39 bytes (7%) of dynamic memory, leaving 473 bytes for local variables. Maximum is 512 bytes.

Thanks Eddy

`#include <Stepper.h>  // Include the Arduino Stepper.h library

// ** Utility function only used when testing in simulator so not needed in actual application **
// size_t
// check_memory()
// {
//   char*   heap_pointer{reinterpret_cast<char*>(malloc(1))};
//   free(heap_pointer);
//   char*   stack_pointer{reinterpret_cast<char*>(&heap_pointer)};
//   return stack_pointer - heap_pointer;
// }

struct Pin_Change
{
  uint8_t const  pin_number;
  uint8_t const  new_value;
};

struct Pin_Changes
{
  size_t const      number_of_pins_to_change;
  Pin_Change const  changes_to_make[NUM_DIGITAL_PINS];
};

enum Step_Direction {step_inwards = -1, step_outwards = 1};

/*************************************
 * Configure the limit switch inputs *
 ************************************/
constexpr uint8_t  home_limit_switch{7};
constexpr uint8_t  max_travel_limit_switch{6};

/****************************
 * Define the stepper motor *
 ***************************/
constexpr uint8_t  motor_pins[] = {8, 10, 9, 11};
Stepper            stepper_motor = Stepper{2048,
                                           motor_pins[0],
                                           motor_pins[1],
                                           motor_pins[2],
                                           motor_pins[3]};
constexpr long  default_step_speed{5};

void
setup()
{
  /*************************************
   * Configure the limit switch inputs *
   ************************************/
  pinMode(home_limit_switch, INPUT_PULLUP);
  pinMode(max_travel_limit_switch, INPUT_PULLUP);

  /********************
   * Configure ouputs *
   *******************/
  digitalWrite(2, LOW);
  pinMode(2, OUTPUT);
  digitalWrite(3, LOW);
  pinMode(3, OUTPUT);
  digitalWrite(4, LOW);
  pinMode(4, OUTPUT);
  digitalWrite(5, LOW);
  pinMode(5, OUTPUT);

  /*****************************
   * Run the movement sequence *
   ****************************/
  stepper_motor.setSpeed(default_step_speed);
  step_until_limit(stepper_motor, step_inwards, home_limit_switch);

  make_moves(stepper_motor);

  stepper_motor.setSpeed(default_step_speed);
  step_until_limit(stepper_motor, step_inwards, home_limit_switch);

  power_off_stepper(motor_pins);
}

void
loop()
{}

void
step_until_limit(Stepper&            stepper_motor,
                 unsigned int const  step_direction,
                 uint8_t const       travel_limit_input)
{
  while (HIGH == digitalRead(travel_limit_input))
  {
    stepper_motor.step(step_direction);
  }
}

void
make_moves(Stepper&  stepper_motor)
{
  // Movement 1
  //  - Outward 600 steps, speed 15
  move_steps_with_stop_limit(stepper_motor, step_outwards, 600, 15, max_travel_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 2 sec pause
  delay(2000);

  // Movement 2
  //  - 2 pin changes
  change_pins({2, {{3, HIGH}, {2, LOW}}});
  //  - Outward 700 steps, speed 3
  move_steps_with_stop_limit(stepper_motor, step_outwards, 700, 3, max_travel_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 2.5 sec pause
  delay(2500);

  // Movement 3
  //  - 3 pin changes
  change_pins({3, {{4, HIGH}, {3, LOW}, {2, LOW}}});
  //  - Inward 600 steps, speed 10
  move_steps_with_stop_limit(stepper_motor, step_inwards, 600, 10, home_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 5 sec pause
  delay(5000);

  // Movement 4
  //  - 2 pin changes
  change_pins({2, {{3, HIGH}, {2, LOW}}});
  move_steps_with_stop_limit(stepper_motor, step_inwards, 600, 10, home_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 0.75 sec pause
  delay(2000);

  // Movement 5
  //  - 4 pin changes
  change_pins({4, {{5, HIGH}, {4, LOW}, {3, LOW}, {2, LOW}}});
  //  - Inward 500 steps, speed 10
  move_steps_with_stop_limit(stepper_motor, step_inwards, 500, 10, home_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 2.25 sec pause
  delay(2250);

  // Movement 6
  //  - 2 pin changes
  change_pins({2, {{3, HIGH}, {2, LOW}}});
  //  - Inward 700 steps, speed 10
  move_steps_with_stop_limit(stepper_motor, step_inwards, 700, 10, home_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 2.5 sec pause
  delay(2500);

  // Movement 7
  //  - 3 pin changes
  change_pins({3, {{4, HIGH}, {3, LOW}, {2, LOW}}});
  //  - Outward 500 steps, speed 8
  move_steps_with_stop_limit(stepper_motor, step_outwards, 500, 8, max_travel_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 3.75 sec pause
  delay(3750);

  // Movement 8
  //  - 2 pin changes
  change_pins({2, {{3, HIGH}, {2, LOW}}});
  //  - Inward 800 steps, speed 15
  move_steps_with_stop_limit(stepper_motor, step_inwards, 800, 15, home_limit_switch);
  //  - 1 pin change
  change_pins({1, {{2, HIGH}}});
  //  - 2 sec pause
  delay(2000);
}

void
change_pins(Pin_Changes const  pin_changes)
{
  size_t  number_of_changes{pin_changes.number_of_pins_to_change};

  if (NUM_DIGITAL_PINS < pin_changes.number_of_pins_to_change)
  {
    number_of_changes = NUM_DIGITAL_PINS;
  }

  while(0 < number_of_changes)
  {
    --number_of_changes;
    digitalWrite(pin_changes.changes_to_make[number_of_changes].pin_number,
                 pin_changes.changes_to_make[number_of_changes].new_value);
  }
}

void
move_steps_with_stop_limit(Stepper&              stepper_motor,
                           Step_Direction const  step_direction,
                           size_t                steps_to_move,
                           long const            speed_to_move_at,
                           uint8_t const         travel_limit_input)
{
  if ((0 == step_direction) || (0 == steps_to_move) || (0 == speed_to_move_at))
  {
    return;
  }

  stepper_motor.setSpeed(speed_to_move_at);

  while ((0 < steps_to_move) &&
         (HIGH == digitalRead(travel_limit_input)))
  {
    stepper_motor.step(step_direction);
    --steps_to_move;
  }
}

void
power_off_stepper(uint8_t const  motor_pins[4])
{
  digitalWrite(motor_pins[0], LOW);
  digitalWrite(motor_pins[1], LOW);
  digitalWrite(motor_pins[2], LOW);
  digitalWrite(motor_pins[3], LOW);
}
`

founfabug, 2023-01-17:
I find ATTiny85 units fail in this way when memory is too tight. They seem to execute an arbitrary piece of code as if in a loop. I tested by adding and removing strings and it was repeatable.

SpenceKonde, 2023-03-17:
Nothing in there looks remotely ram intensive, and flash usage can go all the way to 100%. 

So, no I would stay that it's absolutely not running out of memory. I would be suspicious of noise that the motors are spewing onto the power rails causing resets, just judging by hoe many times I see the word motor.

SpenceKonde, 2023-03-30:
It's strange that the result is a loop, Because one of two things will happen: Either malloc will return a null pointer which the  user code doesn't check for and hence smashes the low I/O registers,. Or it will check it once per loop, get null back and wait untiil next loop, to try again, but the needed memory will never be freed....

Or it could happen on the other half - heap has grown to nearly touch the stack, stack expands, smashing into the heap. Typically this results on both areas being corrupted, and stack corruption is what really puts you in bad territory, because the first thing you trash is the address that you're supposed to go back to on Return/RETI, leading to jumping to arbitrary parts of the program...