SpenceKonde/ATTinyCore #746: Won't compile sketch on ATtiny85 when "transmit only" is set. TinySoftwareSerial error.

nodemand, 2023-02-01:
Sketches won't compile on ATtiny85 when I set the TinySoftwareSerial flag/fuse to "transmit only". 
My setup looks like this:

`
  Serial.setTxBit(3);
  Serial.begin(9600);
`

`
Using board 'attinyx5' from platform in folder: /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr
Using core 'tiny' from platform in folder: /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr
Detecting libraries used...
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/sketch/pitch-shift-05.ino.cpp -o /dev/null
Generating function prototypes...
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/sketch/pitch-shift-05.ino.cpp -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/jeroen/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/sketch/pitch-shift-05.ino.cpp -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/sketch/pitch-shift-05.ino.cpp.o
Compiling libraries...
Compiling core...
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/HardwareSerial.cpp -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/core/HardwareSerial.cpp.o
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/Tone.cpp -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/core/Tone.cpp.o
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -c -g -Os -Wall -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mrelax -Werror=implicit-function-declaration -Wundef -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/WInterrupts.c -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/core/WInterrupts.c.o
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/Serial1.cpp -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/core/Serial1.cpp.o
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10607 -DARDUINO=10607 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny -I/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/variants/tinyx5 /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/TinySoftwareSerial.cpp -o /private/var/folders/75/mscbzkzs4n9bh3qx1rjntwmm0000gn/T/arduino-sketch-3996AAE2AE6D0776E01D3C46A43BEF0B/core/TinySoftwareSerial.cpp.o
/Users/jeroen/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOF [... cut ...]
In file included from /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/Arduino.h:567:0,
                 from /Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/TinySoftwareSerial.cpp:28:
/Users/jeroen/Dropbox/Arduino/hardware/ATTinyCore-2.0.0/avr/cores/tiny/TinySoftwareSerial.h:146:29: note: previous declaration 'TinySoftwareSerial Serial'
   extern TinySoftwareSerial Serial;
                             ^~~~~~

exit status 1

Compilation error: exit status 1
`

hmeijdam, 2023-02-01:
I get the same error, when compiling below in Github V2.0 dev of ATTinyCore with option "built in SoftwareSerial transmit only"
```
void setup() {
Serial.begin(19200);
}

void loop() {
Serial.println("test");
delay(500);
}
```

error

```
Arduino: 1.8.13 (Windows 10), Board: "ATtiny85/45/25 (No Bootloader), -Os (size, recommended), ATtiny85, 8 MHz (internal), EEPROM retained, No receiving, transmit only., B.O.D. Disabled (saves power), Enabled"

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:optimization=size,chip=85,clock=internal_8m,eesave=aenable,softserial=txonly,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_955545 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_667572 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_886885\sketch_feb01a.ino

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -compile -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:optimization=size,chip=85,clock=internal_8m,eesave=aenable,softserial=txonly,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_955545 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_667572 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_886885\sketch_feb01a.ino

Using board 'attinyx5' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Using core 'tiny' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Detecting libraries used...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_955545\\sketch\\sketch_feb01a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_955545\\sketch\\sketch_feb01a.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_955545\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

"E:\\Arduino\\arduino-1.8.13-attinycore\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_955545\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp [... cut ...]
In file included from E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny\Arduino.h:567:0,

                 from E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny\TinySoftwareSerial.cpp:28:

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny\TinySoftwareSerial.h:146:29: note: previous declaration 'TinySoftwareSerial Serial'

   extern TinySoftwareSerial Serial;

                             ^~~~~~
exit status 1

Error compiling for board ATtiny85/45/25 (No Bootloader).

```

SpenceKonde, 2023-02-01:
Should be fixed now

hmeijdam, 2023-02-01:
Something has changed, because now I get a different error

```
Arduino: 1.8.13 (Windows 10), Board: "ATtiny85/45/25 (No Bootloader), -Os (size, recommended), ATtiny85, 8 MHz (internal), EEPROM retained, No receiving, transmit only., B.O.D. Disabled (saves power), Enabled"

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:optimization=size,chip=85,clock=internal_8m,eesave=aenable,softserial=txonly,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_273140 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_753444 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_669124\sketch_feb01a.ino

E:\Arduino\arduino-1.8.13-attinycore\arduino-builder -compile -logger=machine -hardware E:\Arduino\arduino-1.8.13-attinycore\hardware -hardware E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13-attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13-attinycore\hardware\tools\avr -built-in-libraries E:\Arduino\arduino-1.8.13-attinycore\libraries -libraries E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attinyx5:optimization=size,chip=85,clock=internal_8m,eesave=aenable,softserial=txonly,bod=disable,millis=enabled -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_273140 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_753444 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_669124\sketch_feb01a.ino

Using board 'attinyx5' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Using core 'tiny' from platform in folder: E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr

Detecting libraries used...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_273140\\sketch\\sketch_feb01a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_273140\\sketch\\sketch_feb01a.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_273140\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

"E:\\Arduino\\arduino-1.8.13-attinycore\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_273140\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"E:\\Arduino\\arduino-1.8.13-attinycore\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny85 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DSOFT_TX_ONLY -DARDUINO=10813 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX5 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\cores\\tiny" "-IE:\\Arduino\\arduino-1.8.13-attinycore\\portable\\sketchbook\\hardware\\ATTinyCore\\avr\\variants\\tinyx5" "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_273140\\sketch\\sketch_feb01a.ino.cpp" -o "C:\\Users\\Hans\\AppData\\Local\\Temp\\arduino_build_273140\\sketch\\sketch_feb01a.ino.cpp.o"

[... cut ...]
In file included from E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny\Arduino.h:567:0,

                 from E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny\TinySoftwareSerial.cpp:28:

E:\Arduino\arduino-1.8.13-attinycore\portable\sketchbook\hardware\ATTinyCore\avr\cores\tiny\TinySoftwareSerial.h:146:29: note: previous declaration 'TinySoftwareSerial Serial'

   extern TinySoftwareSerial Serial;

                             ^~~~~~

exit status 1

Error compiling for board ATtiny85/45/25 (No Bootloader).

```

commit 27b31dc3 (2023-02-01): Should fix #746

Couple of other things going in here - but the silly errors in the soft serial were trivial to fix. I'd forgotten to #idfef some stuff out
commit c32e88f4 (2023-03-17): May fix #746
closed 2023-03-17 by SpenceKonde