SpenceKonde/ATTinyCore #747: Compile suddenly quit working for ATTiny88

lesrhorer, 2023-03-05:
I have chosen an ATTiny88 for both economic reasons and because it has plenty of Analog pins.  Progress was steady until 2 days ago when suddenly the code would no longer compile under Windows, even on code that had been working perfectly well.  Apparently, the avr-g++.exe file has gone missing both from the ATTinyCore board compiler and the Digistump compiler.  The code will compile under Linux, but I am having some issues uploading the compiled binary to the board under Linux.  This is the error I get when compiling ATTinyCore:

`Arduino: 1.8.13 (Windows 7), Board: "ATtiny167/87 (Optiboot), Enabled, ATtiny167, 8 MHz (internal), B.O.D. Disabled (saves power), New style (down each side), Enabled, Port A (see chart)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lrhorer\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lrhorer\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lrhorer\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attinyx7opti:LTO=enable,chip=167,clock=8internal,bod=disable,pinmapping=new,millis=enabled,neopixelport=porta -ide-version=10813 -build-path C:\Users\lrhorer\AppData\Local\Temp\arduino_build_577229 -warnings=none -build-cache C:\Users\lrhorer\AppData\Local\Temp\arduino_cache_647126 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b -prefs=runtime.tools.micronucleus-2.5-azd1b.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b -prefs=runtime.tools.avrdude.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avr-gcc.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose S:\Equipment\PCBoard Development\Flashlight\Main Board\Arduino_Code\ATTiny88\Buzzer_Test\Buzzer_Test.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lrhorer\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lrhorer\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lrhorer\Documents\Arduino\libraries -fqbn=ATTinyCore:avr:attinyx7opti:LTO=enable,chip=167,clock=8internal,bod=disable,pinmapping=new,millis=enabled,neopixelport=porta -ide-version=10813 -build-path C:\Users\lrhorer\AppData\Local\Temp\arduino_build_577229 -warnings=none -build-cache C:\Users\lrhorer\AppData\Local\Temp\arduino_cache_647126 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b -prefs=runtime.tools.micronucleus-2.5-azd1b.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b -prefs=runtime.tools.avrdude.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avr-gcc.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose S:\Equipment\PCBoard Development\Flashlight\Main Board\Arduino_Code\ATTiny88\Buzzer_Test\Buzzer_Test.ino

Using board 'attinyx7opti' from platform in folder: C:\Users\lrhorer\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2

Using core 'tiny' from platform in folder: C:\Users\lrhorer\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2

Detecting libraries used...

"C:\\Users\\lrhorer\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=8000000L -DCLOCK_SOURCE=0 -DARDUINO=10813 -DARDUINO_AVR_ATTINYX7 -DARDUINO_ARCH_AVR -DNEOPIXELPORT=PORTA "-IC:\\Users\\lrhorer\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.5.2\\cores\\tiny" "-IC:\\Users\\lrhorer\\AppData\\Local\\Arduino15\\packages\\ATTinyCore\\hardware\\avr\\1.5.2\\variants\\tinyX7_New" "C:\\Users\\lrhorer\\AppData\\Local\\Temp\\arduino_build_577229\\sketch\\Buzzer_Test.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

exec: "C:\\Users\\lrhorer\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++": file does not exist

Error compiling for board ATtiny167/87 (Optiboot).
`
I get essentially the same error if I choose the DigiStump compiler, except the directory is:

`exec: "C:\\Users\\lrhorer\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++": file does not exist
`

SpenceKonde, 2023-03-05:
Is the file present? C:\Users\lrhorer\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin\avr-g++

lesrhorer, 2023-03-05:
No, it is not.  While it is possible this  could be the case, I cleared out everything and reinstalled, but the files were still missing.  I then did release three files with potential impact from quarantine, but now the compile cannot find stdlib.h:
`C:\Users\lrhorer\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.7.4\cores\tiny88/Arduino.h:4:10: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
`

lesrhorer, 2023-03-05:
Obviously, I need to get this resolved, but until I do, I do have a compiled binary from the Linux version of the IDE.  How can I upload the binary into the board for testing so I can move forward?

hmeijdam, 2023-03-05:
If I were in your situation, I would add another [portable installation](url) of the Arduino IDE (with virusprotection on hold and on a non-system drive) and try again. That way you keep all required files in one directory tree and you have an isolated environment to try things.

lesrhorer, 2023-03-06:
I wiped everything (again) and installed the IDE (again), and now it compiles without error in Windows 7.  I cannot get it to upload in either Linux or Windows, however.  In Windows the USB driver loads without error and claims the ports are ready, but the virtual port does not show up in the Device Manager. and the IDE times out when attempting to upload.

hmeijdam, 2023-03-06:
Nu clue why the link was wrong, but it was. Here is the right one

https://docs.arduino.cc/software/ide-v1/tutorials/PortableIDE

SpenceKonde, 2023-06-20:
This is a system/IDE issue, not a core issue. It cannot be resolved modification of the core, you must figure out what is wrong with your system and correct it, and then things should work normally - this is beyond the scope of the core, please refer to the arduino.cc installation and troubleshooting forums or other resources for additional help.