SpenceKonde/ATTinyCore #748: ATtiny841/441 error on v2.0.0-dev when using analog and digital ports with CCW mapping

caiser01, 2023-03-05:
Using v2.0.0-dev, if one tries to build the simple example below for the ATtiny841/441 with the pin mapping set to 'Legacy (counterclockwise)'...

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
  
  analogRead(A0);
}
```

...one gets the follwing build errors:

```
Arduino: 1.8.19 (Windows 10), Board: "ATtiny841/441 w/Optiboot serial bootloader, -Os (size, recommended), ATtiny841, 8 MHz (internal > 4.5V), Legacy (counterclockwise), Master Only, UART0: TX PA1, RX PA2. SPI: SCK PA4, MISO PA5, MOSI PA6, SS PA7, B.O.D. Enabled (1.8v), B.O.D. Disabled (saves power), B.O.D. Disabled (saves power), Enabled, Standard (1s wait, for use w/autoreset), Serial 0 (TX: PA1, RX: PA2, LED: PB2)"

In file included from C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny\Arduino.h:230:0,

                 from C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny\wiring_private.h:35,

                 from C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny\wiring_digital.c:11:

C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\variants\tinyx41_legacy/pins_arduino.h:348:2: warning: #warning "This is the COUNTERCLOCKWISE pin mapping - make sure you're using the pinout diagram with the pins in counter clockwise order" [-Wcpp]

 #warning "This is the COUNTERCLOCKWISE pin mapping - make sure you're using the pinout diagram with the pins in counter clockwise order"

  ^~~~~~~

C:\Users\redacteduser\AppData\Local\Temp\ccyLeMll.ltrans0.ltrans.o: In function `digitalWrite.constprop.0':

C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny/wiring_digital.c:153: undefined reference to `port_to_pullup_PGM'

C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny/wiring_digital.c:153: undefined reference to `port_to_pullup_PGM'

C:\Users\redacteduser\AppData\Local\Temp\ccyLeMll.ltrans0.ltrans.o: In function `pinMode':

C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny/wiring_digital.c:29: undefined reference to `port_to_pullup_PGM'

C:\Users\redacteduser\Documents\Arduino\hardware\ATTinyCore-2.0.0-dev\avr\cores\tiny/wiring_digital.c:29: undefined reference to `port_to_pullup_PGM'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board ATtiny841/441 w/Optiboot serial bootloader.

```

Changing the pin mapping to 'Standard (clockwise)' allows the code to compile without issue.

With the pin mapping set to 'Legacy (counterclockwise)', you can also make the code compile by commenting out the pinMode and DigitalWrite lines OR the analogRead line.

I've tried both the 'No Bootloader' and Optiboot variants of the ATtiny841/441 and the behavior is the same. I also spot checked a couple of other devices that support changing the pin mapping such as the ATtiny84 and they did not exhibit this issue so it may be specific to the ATtiny841/441.

SpenceKonde, 2023-03-05:
Thanks, is a trivial thing to fix.

SpenceKonde, 2023-03-17:
Should be fixed in github now.

commit 47fd22d5 (2023-03-17): Fix #748

Thanks, the port-to-pullup table was missing.
closed 2023-03-17 by SpenceKonde