SpenceKonde/ATTinyCore #749: To adopt avrdude 7.1

mcuee, 2023-03-13:
It would be good to adopt avrdude 7.1 release in the future version. 

FYI: MicroCore has adopted avrdude 7.1
https://github.com/MCUdude/MicroCore/commit/0d4e69a33efc21a23378f7adf76f2fa5f518eb6d

mcuee, 2023-03-16:
Binary release from Arduino avrdude-packing project.
https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

MCUdude, 2023-04-05:
Avrdude 7.1 also has support for the Micronucleus bootloader, which means that with 7.1 you can (again) use Avrdude to deal with everything upload related.

This is a good thing because it means that you don't have to handle chips running the Micronucleus bootloader with a separate tool.

It would be cool if ATtinyCore 2.0.0 used Avrdude 7.1, Urboot instead of Optiboot, and Avrdude instead of Micronucleus.

mcuee, 2023-10-05:
Arduino avrdude-packing project has released their version of Arduino 7.2, which should be suitable for the usage here.
https://github.com/arduino/avrdude-packing/releases/tag/7.2-arduino.1

mcuee, 2024-02-16:
avrdude 7.3 has been released.
https://github.com/avrdudes/avrdude/releases/tag/v7.3