SpenceKonde/ATTinyCore #752: Downloading the index failed - please host it in GitHub

DeflateAwning, 2023-03-18:
Downloading the index from the normal URL (http://drazzy.com/package_drazzy.com_index.json) failed. Also, it's only http (not https). Can we add that file to this repo, and then update the links to pull it from the repo in raw format instead?

SpenceKonde, 2023-03-18:
The page is loading fine for me. I suspect transient network connectivity issue (likely on your side). TBPH, my site is more reliably available than stuff hosted through github lately it seems. Github is awesome and all, but it's got so much stuff going on on it that we occasionally get buffeted by the headwinds of what others are doing on the platform.

But as regards http vs https, there is already a plan in motion there - Soon we will have the json file mirrored in the root of azduino.com as well (Can you believe it? Nobody had taken Azduino.com!). and that system DOES support 

I am *not* going to host the json index on github. What kind of scrub do you think I am?! You think I can't get a real https webserver? 

At least at some point, we were advised that it should be located in the root of the webserver virtual file system, and we have our own domain name and webserver with https. Hence why it's located precisely where it is. 

But Azduino.com is where it belongs. 

It seems like the URL can change arbitrarily - but the file **name** was set in stone the moment it was first used; nothing but literal divine intervention could prevent untold woe and misery from being inflicted upon our users if we changed it (And He would have to run time back several years,  to when the file was first used, change the name, then run time forwards to return to the present day. I'm sure He in his infinite wisdom, can think of much better things to do if he was going rewind time a few years in order to change something.  Or maybe He could just use divinegrep, the regex tool God has that will perform regex replacement on the memory of all computing devices, regardless of where they are, who controls them, whether they're air gapped, or even powered on, plus (with the right command line switches) edit any printed or handwritten representations of it, any and all applicable memories in the minds of sentient beings and modify the essential nature of any relevant qualia (I'm not sure if there's a God (there is an absence of evidence in either direction here). But if there is one, I imagine he would have access to universe scopeed versions of the common posix command line tools. He'd have to, or he stopped being omnipotent a few decades ago as Google and Yahoo search were getting started. Since the nature of God is inherently omnipotent and eternal, he could not have ceased to be omnipotent, and hence any being claiming to be such an entity must have a means tobring our technology to heel. That's the most plausible approach (it's common knowledge that the universe runs on Quantium Unix, operrating on the spacetime fabric, with an infinite number of asynchronous cores each operating at F_CPU = 1.8 x 10<sup>43</sup> Hz in the local frame of reference - so all-powerful POSIX commands would seem to His natural tools of choice for changing reality to suit his will). 
But I digress. The point is we can change the URL of the package index file, but it can NEVER BE RENAMED. 

So.....  The plan is that while the current location will be maintained, we're going to stop giving out this URL and start giving out a new URL on the new domain, Azduino.com. which has HTTPS set up..The process for this is a little more complicated than you'd think though, as I need to get the syncing set up on the new server, (otehr than that, it's going to mostly be redicrtects. I think what I *really* need to do is rebuild the server underlying azduino.com. Then cut over to the new one and start giving the new URL. Then I need to deploy what will be the drazzy.com webserver eventually, get it doing the things drazzy,com does now - only on  a newer AMI that can be hosted less expensively. And at that point DrAzzy.com may or may not get optional HTTPS (If I can make that work - the goal is that drazzy,com will *NOT* *require* HTTPS for anything, whereas I do not make the same claim about Azduino.com. DrAzzy,com is where I will put things that need to be talked to by embedded devices that can't handle SSL, which is most of them. 

There is a change brewing here.... that magic json file is going to move from (well, it will still be mirrored there) drazzy.com to azduino.com. Note that while we can change the URL, we cannot change the name of the file, **because that is how the IDE identifies these platfiorm files. I am not joking**.

SpenceKonde, 2023-03-18:
Closing as this is not relevant to this core in particular and is already part of ongoing migration plans.

closed 2023-03-18 by SpenceKonde

DeflateAwning, 2023-03-19:
Awesome, glad you have it figured out.

Link works fine now.

I tried to download from several VPSs/VPNs, and on local residential internet. The issue was not transient on my side. 

10^43 GHz? What?

SpenceKonde, 2023-03-19:
Very strange indeed. 

1.8 x 10^43 Hz,  1.8 x 10^34 GHz 

That's 1 over the plank time, which is the natural unit for time at fundamental physics calculations (such that all the constants in the equations go to unity, greatly simplifying the mathematics), the time that it takes a photon to traverse 1 plank length  (which is quite a small distance). It is known that the standard model of physics is invalid for analysis of processes occurring at the plank scale (where length and time units are tiny relative to our day-to-day experience, and energy and temperature unimaginably high - "immediately after big bang" or "inside of a black hole"). Depending on which physicist you ask, some will posit that in the ultimate description of reality, time and space themselves are quantized, likely around the plank scale; this would be  the natural candidate for the quanta of time. 

So it stands to reason that in this giant quantum computer in which all of existence resides, and within which we all appear to be confined [the universe is clearly, writ large, a quantum computer. Whether it came to be through means natural or through the act of some being from beyond, I couldn't say (ie, I have no position on the question of whether the universe is a simulation), only that if it is a simulation, it's running on an extraordinarily large quantum computer). But either way, the laws of physics underlying everything deep enough down are quantum, hence the universe either is or is running on a quantum computer], since the clock speed of a computer simulating something must be no slower than the briefest meaningful interval of time, so therefor, each "processor" (of which there would have to be infinitely many, as the universe to the best of our knowledge appears to be of infinite size) must execute one cycle every plank time... hence it's operating frequency would be 1.8x10^43 Hz (assuming it is in the same frame of reference as the observer and is not subject to unusually strong gravitational fields,, such as that of a black hole, as those would cause time dilation....))