SpenceKonde/ATTinyCore #753: ATtiny85 error in Arduino.h

ilisk97, 2023-03-22:
I am writing a program for a pair of time-of-flight sensors feeding data into an ATtiny85. The old core I was using kept returning compiler errors and conflicts with TinyWire. So I tossed the old core and installed 2.0.0 of this one since I can just use the normal Wire library. However, I am still receiving errors, and I believe it's an issue with the core (although I'm no expert so I could be wrong).

There are three libraries I am using in the program: <Wire.h>, <SoftwareSerial.h>, and the TOF sensor library <Adafruit_VL6180X.h>. The board selected is "ATtiny85/45/25 (No Bootloader)" because I am using my uno as a programmer. The chip has been set to "ATtiny85," programmer is "Arduino as ISP," and I've got the clock set to 1MHz. The error keeps referencing an "endif without if" in the Arduino.h file contained within avr\cores\tiny (note that the address actually reads \avr\cores\tiny/Arduino.h in case that forward slash is the issue).

I wanted to make sure the issue was with the core and not anything else, so I removed all the libraries and basically uploaded a bare minimum sketch to the ATtiny85. The issues with the adafruit libraries went away (mainly because they too call Arduino.h), but the endif without if error was still present in the core files. Is this a problem with the core or am I missing something?

**Here is the code I am uploading:**
`
#include <Wire.h>
#include <Adafruit_VL6180X.h>
#include <SoftwareSerial.h>

const int pinRx = 3;
const int pinTx = 4;

Adafruit_VL6180X vl = Adafruit_VL6180X();
SoftwareSerial mySerial(pinRx, pinTx);

void setup(){
  pinMode(pinRx, INPUT);
  pinMode(pinTx, OUTPUT);
  Wire.begin();
  mySerial.begin(115200);
}

void loop(){
  float lux = vl.readLux(VL6180X_ALS_GAIN_5);
  uint8_t range = vl.readRange();
  uint8_t status = vl.readRangeStatus();

  if(status == VL6180X_ERROR_NONE) {
    mySerial.print("Range = "); mySerial.println(range);
  }
  else {
    mySerial.println("error");
  }
  delay(50);
}`

**Here is the error it returned:**

`In file included from C:\Users\ian.lisk\AppData\Local\Temp\arduino\sketches\65A6B46DA9417B3D275ABEEAF320C06B\sketch\Zeta-I.ino.cpp:1:0:
C:\Users\ian.lisk\Documents\Arduino\hardware\ATTinyCore-2.0.0\avr\cores\tiny/Arduino.h:89:2: error: #endif without #if
 #endif
  ^~~~~
In file included from C:\Users\ian.lisk\AppData\Local\Temp\arduino\sketches\65A6B46DA9417B3D275ABEEAF320C06B\sketch\Zeta-I.ino.cpp:1:0:
C:\Users\ian.lisk\Documents\Arduino\hardware\ATTinyCore-2.0.0\avr\cores\tiny/Arduino.h:761:2: error: #endif without #if
 #endif
  ^~~~~
In file included from c:\Users\ian.lisk\Documents\Arduino\libraries\Adafruit_VL6180X-master/Adafruit_VL6180X.h:24:0,
                 from C:\Users\ian.lisk\Documents\Arduino\Zeta-I\Zeta-I.ino:6:
C:\Users\ian.lisk\Documents\Arduino\hardware\ATTinyCore-2.0.0\avr\cores\tiny/Arduino.h:89:2: error: #endif without #if
 #endif
  ^~~~~
In file included from c:\Users\ian.lisk\Documents\Arduino\libraries\Adafruit_VL6180X-master/Adafruit_VL6180X.h:24:0,
                 from C:\Users\ian.lisk\Documents\Arduino\Zeta-I\Zeta-I.ino:6:
C:\Users\ian.lisk\Documents\Arduino\hardware\ATTinyCore-2.0.0\avr\cores\tiny/Arduino.h:761:2: error: #endif without #if
 #endif
  ^~~~~
In file included from c:\Users\ian.lisk\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_I2CDevice.h:4:0,
                 from c:\Users\ian.lisk\Documents\Arduino\libraries\Adafruit_VL6180X-master/Adafruit_VL6180X.h:25,
                 from C:\Users\ian.lisk\Documents\Arduino\Zeta-I\Zeta-I.ino:6:
C:\Users\ian.lisk\Documents\Arduino\hardware\ATTinyCore-2.0.0\avr\cores\tiny/Arduino.h:89:2: error: #endif without #if
 #endif
  ^~~~~
In file included from c:\Users\ian.lisk\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_I2CDevice.h:4:0,
                 from c:\Users\ian.lisk\Documents\Arduino\libraries\Adafruit_VL6180X-master/Adafruit_VL6180X.h:25,
                 from C:\Users\ian.lisk\Documents\Arduino\Zeta-I\Zeta-I.ino:6:
C:\Users\ian.lisk\Documents\Arduino\hardware\ATTinyCore-2.0.0\avr\cores\tiny/Arduino.h:761:2: error: #endif without #if
 #endif
  ^~~~~

exit status 1

Compilation error: exit status 1`

SpenceKonde, 2023-03-22:
Thanks, I'll trytop get that fixed soon.

SpenceKonde, 2023-03-22:
Should be fixed by 4f8d018

closed 2023-03-22 by SpenceKonde

sleemanj, 2023-03-22:
bagCall(...)Typo?  In a few places. ---- On Thu, 23 Mar 2023 12:30:44 +1300  ***@***.***  wrote ----
Should be fixed by 4f8d018

SpenceKonde, 2023-03-23:
oops