SpenceKonde/ATTinyCore #754: Recurring issues like in #753

ilisk97, 2023-03-24:
Hello, it's me again from issue #753 

I just tried to compile my program on an ATtiny85 (I am using 09e0cfb) and I got a new error messages which I was able to correct:

1. avr/variants/tinyx5/pins_arduino_.h was renamed to pins_arduino.h
2. Replaced if{} with if/endif on line 129 of the same pins_arduino.h file.

I am now getting the same "error: endif without if" message that I have no idea how to fix. It's on lines 274:2 and 275:2 of pins_arduino.h for the tinyx5 series. Hope this helps!

-Ian

SpenceKonde, 2023-03-24:
https://github.com/SpenceKonde/ATTinyCore/actions/runs/4511658635/

See the Compile Examples jobs.....

ilisk97, 2023-03-24:
I don't follow, is the solution to the issue there? I'm still rather new to github beyond just downloading libraries, not sure what I'm looking at.

SpenceKonde, 2023-03-25:
More recent run now that I've fixed the issue that prevented the name of the part from being shown on the left. Parts with a red X have one or more automated compile tests failing currently. 

https://github.com/SpenceKonde/ATTinyCore/actions/runs/4515681343

commit 535d7e06 (2023-03-26): Fix #754  and similar issues for several other devices

Doc formatting corrections, and fix several variant issues

Thiis shojuld get the number of CI failures way down.

Also fix the CI to not fail to put additional parameters in the names of runs.
closed 2023-03-26 by SpenceKonde