SpenceKonde/ATTinyCore #758: Failed to install platform: arduino:mbed_opta

enrob, 2023-04-04:
Hello, previuos week IDE worked fine with Opta. For other reason yesterday i re installed IDE and the result is that :

Error: 13 INTERNAL: Cannot install tool arduino:arm-none-eabi-gcc@7-2017q4: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 5472 != 131761924

Solution ?

Best regards

SpenceKonde, 2023-04-04:
I can only share the first step of a solution:
1. Ask this same question on the github of the core or hardware package you are trying to install, as opposed to what you have done, which is to ask an expert on a different set of parts on a different architecture who has never heard of the package you're trying to install and hasn't compiled anything for arm_eabi in over 5  years. 

I don't know jack-fuck-all about this, because you're asking the wrong hardware package owner. 

Would you go ask a quantum physics professor who speaks only english a question about european literature in the 18th century which you plan to ask in french? Go ask the native french speaking prof in the literature department, who focuses on 18th century europe!

closed 2023-04-04 by SpenceKonde

enrob, 2023-04-04:
Excuse me, i saw after that i posted in wrong section.

Best regards