SpenceKonde/ATTinyCore #760: EEPROM not retained on ATtiny85-20SF but is retained on ATtiny85-20PU and timing issue

krekr, 2023-04-17:
This is driving me nuts. This didn't use to be this way. I also genuinely think I've found a bug. My setup is:
MBP on macOS 12.6.3 (21G419)
Arduino IDE (2.0.4) or Arduino (1.8.19)
ArduinoISP programmer
ATTinyCore 1.4.1 and 1.5.2

When programming an ATtiny85-20SF (SOIC), timings are ok, but the EEPROM is not retained between programming different sketches. I really need that for my application.
BUT, when I insert the DIP version of the same part (ATtiny85-20PU) into the same programmer (on the same machine etc.), the clock is weirdly scaled (e.g. "delay"s are incorrect and I have to select 9600 baud for software serial when 4800 is specified. However, the EEPROM **is** retained!?

Arduino settings:
[image]
Sketches:
[Archive.zip](https://github.com/SpenceKonde/ATTinyCore/files/11251472/Archive.zip)

[This video](https://youtu.be/HbLWDb6Eb2Y) shows the output of a serial pin through separate FTDI into CoolTerm. First the SOIC part is inserted, and at around the 1 minute mark, I swap it for the DIP version. 

I haven't tried the 2.0.0 version since that is in beta.

sleemanj, 2023-04-17:
Most likely you have not set fuses on both of them (burn bootloader, and any options modifying it), and said chips have different fuse settIngs.  Both clocking and EE preservation are fuse settings.Message ID: ***@***.***>

krekr, 2023-04-18:
Oh god, you're right. I should know this stuff. Sorry for wasting your time. The "No bootloader" in the board description made me think I didn't have to "Burn bootloader".

closed 2023-04-18 by krekr