SpenceKonde/ATTinyCore #761: Don't set fuses

drsubs, 2023-04-30:
When I program (ATTiny84) it don't set high and low byte fuse, only when a do a bootloader up load(even though it´s a no bootloader project).

hmeijdam, 2023-04-30:
https://github.com/SpenceKonde/ATTinyCore#when-using-an-individual-chip-for-the-first-time-or-after-changing-the-clock-speed-eesave-or-bod-settings-you-must-do-burn-bootloader-to-set-the-fuses-even-if-you-are-not-using-the-chip-with-a-bootloader

SpenceKonde, 2023-04-30:
This is correct and intended behavior. Unlike modern avrs where we do set
fuses on normal upload (for iirc 6/8ths of fuses that can't make the part
difficult to reprogram without special tools), on classic avrs, 0/3rds of
the fuses can be misconfigured without potentially leaving the part bricked
without special tools (hv programming, which for many parts ATTC supports
means hvpp, with 20 wires to connect correctly). Therefore we follow the
example set by the official arduino core and virtually every third party
core and only set fuses on burn bootloader here.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

drsubs, 2023-04-30:
Thnx

closed 2023-05-01 by SpenceKonde