SpenceKonde/ATTinyCore #762: PWM frequency ATtiny84 (ATtinyx4)

ltwin8, 2023-05-01:
hi, i would like to drive 4 PWM led drivers with an attiny 84 MCU, i noticed i can not get all channels to output at same frequency, the datasheet suggest a prescaler of zero can be applied to both timers from same source, however i can not get it to work.

i would like to achieve a PWM of around 500Hz to avoid visible flicker even on reasonable movements. i like to stay below 1kHz because of vibrations.

TCCR0B = _BV(CS00); //should yielld prescaler = 1 from CLK-i/o
TCCR1B = _BV(CS10); //should yielld prescaler = 1 from CLK-i/o

TCCR0B = TCCR0B | 0b00000001;
TCCR1B = TCCR1B | 0b00000001;
also does not work...

can anyone help me?

SpenceKonde, 2023-05-01:
Prescale = 1 would result in F<sub>pwm</sub> = F_CPU/(TOP+1) = F_CPU/256 = 31250 Hz. 

You can get close to 500 with full 8-bit resolution at 488 Hz (with /64 prescale). In fact, we configure it for 64 prescale by default (we try to stay between 500 and 1500 Hz - much lower than 500 and there's flicker, and above 1500 Hz, even at 5v, you start struggling to drive MOSFETs all the way to on or off fast enough, they spend too long switching, switching losses are high, and gate drivers are needed when you really oughtn't need them - hence that target range). So just doing analogWrite should get you  488 Hz out of one of the two timers (timer0 - we can't use phase correct mode there anyway because we use it for millis, and how to you know if it's going up or down?). You might end up getting half that out of the other timer. That is because it is in phase correct PWM mode, instead of fast PWM mode. (We do the same thing as the official core there, don't look at me!) 
Hence you must change the waveform generation mode (WGM) IIRC the core sets it to 0b0001 (8 bit phase correct) on the non-millis timers, to 0b0101 (8-bit fast PWM). 

I think leaving timer0 as is will give you 488 Hz pwm. Timer 1 will give half that in stock configuration, and getting the same frequency will meanchanging WGM from 0001 to 0101. Conveniently only one bit needs to change, namely bit 3 of TCCR1B

So TCCR1B |= 1<<3; I think should be all you'd need for analog write to get you 488 Hz PWM...

ltwin8, 2023-05-01:
> Prescale = 1 would result in Fpwm = F_CPU/(TOP+1) = F_CPU/256 = 31250 Hz.

Hello, thank you. I do not drive a get directly but low capacitance dimming Input of few pF, faster freq should work also but the resonance of the PCB is around 1-2k (Design constrained)Thanks for your answer, I will give it a tryVon meinem iPhone gesendet

ltwin8, 2023-05-02:
@SpenceKonde just wondering, how can i achieve PWM of similar frequency on attiny 85 pins PB0 and PB1?

i drive a WW/CW LED (CCT light) and use this driver https://www.mouser.de/datasheet/2/277/MP3362GJ-2497282.pdf

i would like to try analog dimming on it but there i have the same problem with the different frequency... can i apply the same rule to attinyX5? the actual mcu silicon is quite similar i think. only dont know if that would mess up something other.... i do not use millis or delay

thanks for the help

SpenceKonde, 2023-05-22:
Oh the tiny85 is a whole other can of worms. The timer1 there is WHACK, It's a high speed timer and it is async. Can be run at 32 or 64 MHz from the PLL. 
If you just need to duplicate the pwm frequencies though, you;re off by a factor of 2,so just nudge the prescaler by 1 - Timer1 gets EVERY POWER OF TWO PRESCALER!

SpenceKonde, 2023-08-13:
This should be solved now. We have a table in each parts part specific documentation that lists the default frequency of PWM at all speeds, I've fixed the prescaler calculations to keep the speeds much more consistent as the clock speed changes when we are able to do that (like, it now switches between phase correct and fast PWM based on F_CPU), since few people care which one it is using (and those that do are taking over the timers anyway)...

closed 2023-08-13 by SpenceKonde