SpenceKonde/ATTinyCore #763: Interrupts on the ATtiny88

electrickery, 2023-05-02:
Based on comments in issue #707 "Interrupt pins on the Attiny88/MH-ET-Tiny88?", I created a minimal sketch demonstrating it works differently from what I expected. 
Expected was a slow blinking of the LED, and extra blinks each time pin 2 is grounded. What appears to happen is that the interrupt happens continuous resulting in very fast blinking, keeping the LED (apparantly) permanent on. But even shorting pin 2 does not stop the interrupts.

Another observation is that the slow blink rate (tested in an other variant of the sketch) is way faster than the defined 1000 ms, probably making the ATtiny88 (no real clock) less suitable for timing dependent applications.

Greetings,

Fred Jan

```
// ATtiny88 interrupt test - Using interrupts with the ATtiny88

// Code extracted from the rs-switch library
// Minimal sketch that should allow control
// of a LED via an interrupt routine.

// Code compiled with Arduino IDE 1.8.189 and 
// ATTinyCore-2.0.0 installed (from zip), 
// exported (Sketch > Export Compiled Binary)
// and programmed using 
// avrdude -c avrisp2 -P usb -p t88 -Uflash:w:'.../sketchbook/tiny88IntTest/tiny88IntTest.ino.attiny88.mhet.8mc0..hex'

#define INT      0
#define INTPIN   2

#define LED      0

int nReceiverInterrupt;
#define RECEIVE_ATTR
 
void setup() {
    pinMode(LED, OUTPUT);
  
    pinMode(INTPIN, INPUT_PULLUP);
    attachInterrupt(INT, handleInterrupt, CHANGE);
}

void loop() {
    digitalWrite(LED, !digitalRead(LED));
    delay(1000L);
}

void RECEIVE_ATTR handleInterrupt() {
  digitalWrite(LED, !digitalRead(LED));
}
```

SpenceKonde, 2023-05-29:
Closing issue; observed behavior is expected when MHET t88 board is plugged into USB port of a computer. 
Workaround: Use INT1. Avoid PD1, PD2. Will amend the part specific documentation for micronucleus boards to highlight this fact,

closed 2023-05-29 by SpenceKonde

electrickery, 2023-05-29:
Hi Spence,
Thanks for your response.
It could be I used the USB hub to power the AT88 board, but indeed when powered from a charger blinking is like it is in the variant sketch (using no interrupts).
Greetings,
Fred Jan