SpenceKonde/ATTinyCore #764: ATTinyCore download CertificateExpiredException

dlkeng, 2023-05-02:
Hi,
I've tried to install ATTinyCore 1.5.2 board in Arduino 1.8.13 on Windows 10 Pro Version 22H2 and am unable to due to error downloading. It appears to be due to "CertificateExpiredException". This also occurs with versions 1.4.0, 1.4.1, and 1.5.1. I am using "http://drazzy.com/package_drazzy.com_index.json" as the Boards Manager link.
Version 1.3.3 can be successfully downloaded and installed.

The following are the errors reported:
```
Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-i686-mingw32.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-i686-mingw32.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-i686-mingw32.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:113)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:965)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379)
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:162)
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:155)
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:106)
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:234)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
	... 5 more
Caused by: sun.security.validator.ValidatorException: PKIX path validation failed: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:362)
	at sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:259)
	at sun.security.validator.Validator.validate(Validator.java:262)
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:324)
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:229)
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:124)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1621)
	... 21 more
Caused by: java.security.cert.CertPathValidatorException: validity check failed
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:135)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:233)
	at sun.security.provider.certpath.PKIXCertPathValidator.validate(PKIXCertPathValidator.java:141)
	at sun.security.provider.certpath.PKIXCertPathValidator.engineValidate(PKIXCertPathValidator.java:80)
	at java.security.cert.CertPathValidator.validate(CertPathValidator.java:292)
	at sun.security.validator.PKIXValidator.doValidate(PKIXValidator.java:357)
	... 27 more
Caused by: java.security.cert.CertificateExpiredException: NotAfter: Sun Apr 30 05:36:53 CDT 2023
	at sun.security.x509.CertificateValidity.valid(CertificateValidity.java:274)
	at sun.security.x509.X509CertImpl.checkValidity(X509CertImpl.java:629)
	at sun.security.provider.certpath.BasicChecker.verifyValidity(BasicChecker.java:190)
	at sun.security.provider.certpath.BasicChecker.check(BasicChecker.java:144)
	at sun.security.provider.certpath.PKIXMasterCertPathValidator.validate(PKIXMasterCertPathValidator.java:125)
	... 32 more
```

SpenceKonde, 2023-05-02:
Good lord LetsEncrypt keeps us on a short leash, and autorenew seems impossible. I swear I just updated the cert. 

It should be fixed now, though your computer may have cached the expired cert

closed 2023-05-02 by SpenceKonde

dlkeng, 2023-05-02:
OK. Was able to successfully update to ATTinyCore 1.5.2 board.

However, the update issues the following warning when it completes:
```
Warning: non trusted contribution, skipping script execution (C:\Users\<username>\AppData\Local\Arduino15\packages\ATTinyCore\tools\micronucleus\2.5-azd1b\post_install.bat)
```

SpenceKonde, 2023-05-04:
Yeah.  run that batch file if you want to use digispark/micronucleus boards and don't have the drivers installed. 

It's ridiculous that they block non-trusted scripts from running.... because like.... it's happy to download a whole 30mb toolchain that I specify, and run the files in it when you compile a sketch - I could totally make a avr-gcc.exe which kicked off a malware install and then forwarded it's arguments to an actual copy of avr-gcc so you wouldn't know anything evil was afoot. The antimalware industry even has a history of getting false positives from arduino compiles because of all the files being created and deleted so people would be inclined to try turning off their AV if it was blocked. It lets me run run a script I supplied to upload and all that jazz. It would be no harder for me to supply malware as the toolchain than to have post_install.bat install malware (actually, it would be easier, because it could reinfect the system every time you compiled). Of course, there doesn't seem to be anyone trying to use Arduino cores as a means of distributing malware - probably because there are only a few people with arduino cores and people tend to know who they are..... But despite how irrational it is, they only run post_install.bat from packages distributed by Arduino.cc. Nobody else is "trusted" enough to let run a post_install script. It's pretty damned stupid. 

I believe this is noted in the installation instructions.