SpenceKonde/ATTinyCore #765: V2.0.0 How to get Micronucleus to work on a portable (Github desktop client) installation?

hmeijdam, 2023-05-04:
I have several portable IDE installations of ATTinycore on the same PC.
In my board manager version IDE 1.8.18 and ATTinycore  1.5.2 uploading via Micronucleus works perfect (digistump pro)
It issues this upload command:
```
E:\Arduino\arduino-1.8.18\portable\packages\ATTinyCore\tools\micronucleus\2.5-azd1b/micronucleus --no-ansi --run --timeout 60 C:\Users\Hans\AppData\Local\Temp\arduino_build_402954/sketch_may04a.ino.hex 
> Please plug in the device (will time out in 60 seconds) ... 
```

In my portable 1.8.13 and (Github desktop synced) AttinyCore V2.0.0 installation it is missing the tools directory where micronucleus.exe resides. It is on the same PC.
I get this error and can't figure out where/how to get that missing tools directory in place. 
```
{runtime.tools.micronucleus.path}/micronucleus --run --dump-progress --timeout 60 C:\Users\Hans\AppData\Local\Temp\arduino_build_34886/sketch_may04a.ino.hex 
java.io.IOException: Cannot run program "{runtime.tools.micronucleus.path}/micronucleus": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
An error occurred while uploading the sketch

```

SpenceKonde, 2023-05-22:
Uh, did you download and install the Micronucleus uploader? It's pulled in using board manager normally and must be manually installed if the core is manually installed.

SpenceKonde, 2023-06-06:
Thinking about it now, I'm not sure how to make it work on a portable installation unless you install it on the portable installation using board manager, because I don't know how to get the {runtime.tools.micronucleus} to resolve to a path, except by installing a board package with board manager. What I do on DxC/mTC for the tools we package is that the manual installation and board manager ones install a different platform.txt, such that the python interpreter used to run the upload tool that uploads to modern AVRs with just a serial adapter has the path set to the usual {runtime.tools....} path, but the manual installation version instead has the location hardcoded to the (architecture)/tools/python-(something) directory, and you can use that version to install there. But if you have a portable installation, why not use board manager on it? It's the non portable ones that should avoid board manager like the plague.

hmeijdam, 2023-06-06:
> But if you have a portable installation, why not use board manager on it?

I would not know how to install your V2.0.0 Github Attinycore version with the board manager, until you release it.

SpenceKonde, 2023-06-06:
oooooooh, 2.0.0 

Install latest (1.5.2 iirc) from board manager, then do manual installation of 2.0.0. (I do not belive the micronucleus version we expect has changed). It go into the (base install location)/portable/sketchbook/hardware folder (hardware will need to be created). You'll end up with 2 copies of the core in boards menu, one will say (in sketchbook), that's the one installed manually, and that's 2.0.0 - that should work.

hmeijdam, 2023-06-06:
I just tried that but in the other order. I added the boards manager 1.5.2 into my manual Github version, but then only the word ...(in sketchbook) dissapears, so no two cores and throwing a compilation error. After removing the hardware folder and leaving the tools folder (with micronucleus.exe), I get the old error again, that it can't find micronucleus.exe

hmeijdam, 2023-06-06:
Ass soon as I add (Github desktop) the V2.0.0 core to a board manager installation of the 1.5.2 core, I will only see the V2.0.0 version of the core and get below compilation error.

```
Arduino: 1.8.13 (Windows 10), Board: "ATtiny88 w/Micronucleus (MH-Tiny/MHET), -Os (size, recommended), 16 MHz (external clock), MH-Tiny, RX on PD7, TX on PD (default PD6), B.O.D. Disabled (saves power), Enabled, External Reset and Power On Reset, MH Tiny - PD1: D-, PD2: D+, PD0: LED, Reset pin. Further ISP programming possible, Upgrade (via USB)"

E:\Arduino\arduino-1.8.13 - attinycore\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\arduino-1.8.13 - attinycore\hardware -hardware E:\Arduino\arduino-1.8.13 - attinycore\portable\packages -hardware E:\Arduino\arduino-1.8.13 - attinycore\portable\sketchbook\hardware -tools E:\Arduino\arduino-1.8.13 - attinycore\tools-builder -tools E:\Arduino\arduino-1.8.13 - attinycore\hardware\tools\avr -tools E:\Arduino\arduino-1.8.13 - attinycore\portable\packages -built-in-libraries E:\Arduino\arduino-1.8.13 - attinycore\libraries -libraries E:\Arduino\arduino-1.8.13 - attinycore\portable\sketchbook\libraries -fqbn=ATTinyCore:avr:attiny88micr:optimization=size,clock=extclk_16m,pinmap=mhet,softserial=enable,bod=disable,millis=enabled,bootentry=extrf_porf,usbpins=standard,resetpin=reset,burnmode=upgrade -ide-version=10813 -build-path C:\Users\Hans\AppData\Local\Temp\arduino_build_526666 -warnings=none -build-cache C:\Users\Hans\AppData\Local\Temp\arduino_cache_854522 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Hans\AppData\Local\Temp\untitled1750896652.tmp\sketch_jun06a\sketch_jun06a.ino

Error resolving FQBN: board
Error compiling for board ATtiny88 w/Micronucleus (MH-Tiny/MHET).

```
After deleting the 1.5.2 version from the "packages" directory the compilation error is gone. So I think it is using the V2.0.0 menu options, it is compiling with the 1.5.2 version.

SpenceKonde, 2023-06-06:
Yeah, it looks like all the tricks I used to use don't work. 

I don't know why. And I can't get that portable installation working at all now. 

The wacky thing is, arduino_debug suggests that the problem is in the logging, which just makes no sense.