SpenceKonde/ATTinyCore #768: I would like to install 2.0.0.0

JohnOH, 2023-05-22:
Thank you for the work you do on cores for the Tiny families. I am currently exploring the T84A and particularly trying to use Serial with TX on an alternate pin. I understand that Serial.setTxBit is only fixed on version 2.0.0.0. I have not been able to see the 2.0.0.0 version using the Board manager on Arduino 1.8.13. Are there any installation intructions specific to the 2.0.0.0 version of ATTinyCore?

Best Regards

SpenceKonde, 2023-05-22:
2.0.0-dev is the development version, it's the current version in github (the name should make it obvious, and it's marked as current head). See: https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/Installation.md

So a standard manual install is the only special step. It's not in board manager because it is not ready for release *points* "Issues: 30"

JohnOH, 2023-05-22:
Thank you for your swift response I am now able to use 2.0.0.0. However, this has brought me to the point I was last year with #672 . Would you prefer a new issue or should I reopen #672

closed 2023-05-22 by JohnOH