SpenceKonde/ATTinyCore #777: Fails to install due to Tool avrdude is not available for your operating system

usr97629238, 2023-06-16:
Fails to install on `linux-i686 (Debian 11), arduino-cli  Version: 0.32.2` due to:
```
arduino-cli core install ATTinyCore:avr
Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Downloading packages...
Error during install: Error downloading tool arduino:avrdude@6.3.0-arduino18: no versions available for the current OS, try contacting packages@arduino.cc
```

Apparently it's the same issue as https://github.com/SpenceKonde/megaTinyCore/issues/557

SpenceKonde, 2023-06-17:
> Tool ***@***.*** already installed
> Error during install: Error downloading tool ***@***.***: no versions available for the current OS, try contacting ***@***.***
> Apparently it's the same issue as SpenceKonde/megaTinyCore#557

I'm aware that we have this issue and it's straightforward to work around
for 2.0.0. Unfortunately the issue cannot be corrected without doing a new
release. Arduino stopped building for 32 bit linux years ago, and this
release was between when they did that, and when I learned that they had
done that. I solved  it by having megaTinyCore and dxcore download an older
version of avrdude for 32-bit linux.  I worked around it when i was able to
do a release, but I don't think I can do a new 1.5.x release at this point
and 2.0.0, the next planned release is 2.0.0. No ETA on that though, as we
don't have enough people testing over there, and it looks like I;m going to
be mostly occupied with the bullshit of moving unexpectedly and against my
will for months, because that's how long it takes to relocate my operation
center.

By 2.0.0 we will be using AVRdude 7.2. If that supports x86 linux, 2.0.0
will, otherwise it will not, and x86 linux as a supported platform will be
dropped from all my cores for uploads through AVRdude (modern AVRs would
still have SerialUPDI

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2023-06-18:
Ah I have just confirmed that it is expected that there will be x86
avrdude7.2 support.

Unfortunately, the head honcho over at avrdude says they only build for
Windows.
That may mean that there will be a windows only release of 2.0.0, followed
by a 2.0.x version that adds support for non-windows platforms once Arduino
has finally created the binaries. This is what I hate about avrdude - it
has to be compiled. And apparently doing this is non-trivial, based on the
observation that everyone seems loath to do it, and is one of the technical
reasons I prefer python based tooling like SerialUPDI. most of the reasons
are aesthetic or ideological.

Mcuee diplomatically said that there might be delays in the arduino build
system binaries due to testing, which surprised me, as I have seen little
evidence that Arduino tests their code prior to release (Sometimes it seems
that microchip doesn't do a great job of testing their hw before they ship
either)

But back onto the main topic: So how do you make the core install under
those conditions?
I'm in the process of a painful webserver rebuild - fallout from the
drazzy.com issue (which has been partly rebuilt - and you'll notice that
https now works too)

The plan though is for there to be a test json file that would have beta
versions under a different name and location on that server, and there, I
could reroll the 1.5.x core with the same core package referencing the
different avrdude package. But until this is done all you can do is manual
install. When I changed the version of avrdude i spec'ed, I was unaware
that support for some platforms had been dropped i was just dutifully
updating to the latest version of the standard tools. Once you put
something into the board index json file, though you must never change it
within that json file or all hell will break loose and it will break
existing installs.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

usr97629238, 2023-06-20:
> Unfortunately, the head honcho over at avrdude says they only build for Windows.

Well, it seems that their idea is that it's available on Linux natively https://packages.debian.org/bookworm/avrdude. However, at least on Debian, that means no major updates until the next Debian release.

SpenceKonde, 2023-06-20:
Because the platform.txt must match the avrdude version. It would be nice if that wasn't the case - but I'm not the one leading the avrdude effort (and if I was, I'd drag it behind the woodshed, and replace it with a few python scripts for the 2-3% of the programmers avrdude supports that are still relevant). The linux package manager supplied version could be anything. Beyond that, I don't trust package managers with Arduino-related stuff due to the well known fact that the Arduino IDE packages themselves if installed via a package manager are typically broken (in ways varying from obvious to subtle), because the package manager people modify them without understanding the impact of the changes they make, in order to conform to their ideology - which is very beautiful and self-consistent, but is not consistent with reality.

closed 2023-06-25 by SpenceKonde