SpenceKonde/ATTinyCore #778: Problems with controlling I/O pins directly on the ATtiny167

technoblogy, 2023-06-18:
I was puzzling about why the following program didn't flash an LED on PB0 on the ATtiny167:
````text
void setup() {
 DDRB = (1<<0);
}

void loop () {
  PORTB = PORTB | (1<<0);
  delay(1000);
  PORTB = PORTB & ~(1<<0);
  delay(1000);
}
````
I finally tracked it down to the fact that the ATTinyCore assigns the PWM pins to Timer/Counters, so they are disabled for direct I/O. I solved it by turning off the PWM outputs on PB0 to PB3 in `setup()` with:

    TCCR1D = TCCR1D & ~(0b00110011);

I've never come up against this problem before with ATTinyCore (or DxCore). Is this something I should be on the lookout for with other processors, or does it only affect the ATtiny87/167?

Wouldn't it be possible to only reassign the I/O pins on the first call to `analogWrite()` to avoid this problem?

Thanks, David

SpenceKonde, 2023-06-18:
That's what I thought I was doing! 

That timer behavior is unique to the timer on the 167, but the behavior intended is for that register bit to be set when analogWrite is called on the pin and cleared when digitalWrite is called on it, and not set any value by default. 

Which version are you seeing this behavior in? It should absolutely not be setting that register on statup

technoblogy, 2023-06-18:
1.5.2 - I think that's the latest?

SpenceKonde, 2023-06-20:
Have you tested with the 2.0.0-dev version? That is where all current developmennt is happening. I feel like I remember writing the code that would fix this, but it's only in 2.0.0-dev versions

technoblogy, 2023-06-20:
OK I'll try that, thanks!

technoblogy, 2023-06-21:
Sorry, same behaviour with 2.0.0-dev version.

ie direct PORTB toggling only works if I do this first:

    TCCR1D = TCCR1D & ~(0b00110011);

SpenceKonde, 2023-06-21:
So wait a minute, you're saying that on startup, TCCR1D is not starting up as 0? What value is it starting up as? 

I searched the source code and don;t see an obvious route towards things going wrong here though it's obviously a bit confusing since there are two parts with a TCCR1D, but the only thing the registers have in common is the name.

technoblogy, 2023-06-21:
Running:
```
void setup() {
  Serial.begin(9600);
  delay(5000);
  Serial.println(TCCR1D);
}

void loop () { }
```
prints 17 (0b00010001).

technoblogy, 2023-06-21:
By the way, I get these warnings when I upload the program (with USBtinyISP - FAST):
````text
avrdude warning: ATtiny841's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:1416]
avrdude warning: ATtiny441's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:1603]
avrdude warning: a8 would normally be expected to be a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2167]
avrdude warning: ATtiny87's eeprom write misses a necessary address bit a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2183]
avrdude warning: ATtiny87's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2183]
avrdude warning: a8 would normally be expected to be a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2354]
avrdude warning: ATtiny167's eeprom write misses a necessary address bit a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2370]
avrdude warning: ATtiny167's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2370]
avrdude warning: ATtiny44's eeprom writepage misses a necessary address bit a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:4405]
avrdude warning: ATtiny84's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:4589]
avrdude warning: a4 would normally be expected to be a5 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude warning: a3 would normally be expected to be a4 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude warning: a2 would normally be expected to be a3 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude warning: a1 would normally be expected to be a2 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude warning: a0 would normally be expected to be a1 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude warning: too few opcode bits in instruction [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude warning: ATtiny43u's eeprom read misses a necessary address bit a5 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5336]
````

SpenceKonde, 2023-06-21:
Well at least you can upload. I can't make any of my upload tools work on a known good tiny841. 
Regarding those warnings, have you independently updated avrdude by any chance? See, I *don't* get those errors...

technoblogy, 2023-06-22:
> have you independently updated avrdude by any chance

No, I just downloaded a .ZIP of 2.0.0-dev and put it into my Arduino/hardware folder. Unless it's picking up avrdude from somewhere else...

SpenceKonde, 2023-06-22:
That has nothing to do with whether, for some other purpose, you may have updated avrdude. For example, you may have installed MegaCoreX through the board manager, which uses AVRdude 7.1. **that may be enough to cause it toi use 7,1 with this** because the 2.0.0-dev wasn't installed by board  manager and hence there is no concept of a "correct toolchain" or "correct avrdude version" , as far as the IDE is aware. So it just uses whatever it's got hanging around. If a board manager has updated one of those, it may use the updated version! (I don't know the algorithm here). verbose output enable during compiling (I do need to make that mandatory here don't I? Though I was reserving my decision until I saw what the 7.1 output was like.

technoblogy, 2023-06-22:
Is there a way to see the path to the avrdude it's using?

SpenceKonde, 2023-06-22:
verbose output while compiling, it will show you the actual commandline it invokes.

technoblogy, 2023-06-22:
I assume you mean verbose output during upload:
````text
avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf
avrdude yywarning() warning: ATtiny841's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:1416]
avrdude yywarning() warning: ATtiny441's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:1603]
avrdude yywarning() warning: a8 would normally be expected to be a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2167]
avrdude yywarning() warning: ATtiny87's eeprom write misses a necessary address bit a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2183]
avrdude yywarning() warning: ATtiny87's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2183]
avrdude yywarning() warning: a8 would normally be expected to be a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2354]
avrdude yywarning() warning: ATtiny167's eeprom write misses a necessary address bit a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2370]
avrdude yywarning() warning: ATtiny167's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:2370]
avrdude yywarning() warning: ATtiny44's eeprom writepage misses a necessary address bit a7 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:4405]
avrdude yywarning() warning: ATtiny84's eeprom writepage misses a necessary address bit a8 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:4589]
avrdude yywarning() warning: a4 would normally be expected to be a5 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude yywarning() warning: a3 would normally be expected to be a4 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude yywarning() warning: a2 would normally be expected to be a3 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude yywarning() warning: a1 would normally be expected to be a2 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude yywarning() warning: a0 would normally be expected to be a1 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude yywarning() warning: too few opcode bits in instruction [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5317]
avrdude yywarning() warning: ATtiny43u's eeprom read misses a necessary address bit a5 [/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/avrdude.conf:5336]
         User configuration file is /Users/david/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 8.0
avrdude usbtiny_open() error: cannot find USBtiny device (0x1781/0xc9f)
avrdude main() error: unable to open programmer usbtiny on port usb

avrdude done.  Thank you.
````

technoblogy, 2023-06-22:
Verbose compiling output:
````text
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/david/Library/Arduino15/packages -hardware /Users/david/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/david/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/david/Documents/Arduino/libraries -fqbn=ATTinyCore-2.0.0-dev:avr:attinyx7:optimization=size,chip=167,clock=internal_8m,pinmap=default,eesave=aenable,bod=disable,millis=enabled -ide-version=10819 -build-path /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /Users/david/Documents/Arduino/TFTGraphicsLibrary2-167/TFTGraphicsLibrary2-167.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/david/Library/Arduino15/packages -hardware /Users/david/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/david/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/david/Documents/Arduino/libraries -fqbn=ATTinyCore-2.0.0-dev:avr:attinyx7:optimization=size,chip=167,clock=internal_8m,pinmap=default,eesave=aenable,bod=disable,millis=enabled -ide-version=10819 -build-path /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /Users/david/Documents/Arduino/TFTGraphicsLibrary2-167/TFTGraphicsLibrary2-167.ino
Using board 'attinyx7' from platform in folder: /Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr
Using core 'tiny' from platform in folder: /Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DARDUINO=10819 -DARDUINO=10819 -DARDUINO_AVR_ATTINYX7 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/cores/tiny -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/variants/tinyx7 /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/sketch/TFTGraphicsLibrary2-167.ino.cpp -o /dev/null
Generating function prototypes...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny167 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DARDUINO=10819 -DARDUINO=10819 -DARDUINO_AVR_ATTINYX7 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/cores/tiny -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/variants/tinyx7 /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/sketch/TFTGraphicsLibrary2-167.ino.cpp -o /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=attiny167 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DARDUINO=10819 -DARDUINO=10819 -DARDUINO_AVR_ATTINYX7 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/cores/tiny -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev/avr/variants/tinyx7 /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/sketch/TFTGraphicsLibrary2-167.ino.cpp -o /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/sketch/TFTGraphicsLibrary2-167.ino.cpp.o
Compiling libraries...
Compiling core...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -x assembler-with-cpp -flto -MMD -mmcu=attiny167 -DF_CPU=8000000UL -DCLOCK_SOURCE=0 -DARDUINO=10819 -DARDUINO=10819 -DARDUINO_AVR_ATTINYX7 -DARDUINO_ARCH_AVR "-DATTINYCORE=\"2.0.0-dev\"" -DATTINYCORE_MAJOR=2UL -DATTINYCORE_MINOR=0UL -DATTINYCORE_PATCH=0UL -DATTINYCORE_RELEASED=0 -I/Users/david/Documents/Arduino/hardware/ATTinyCore-2.0.0-dev [... cut ...] ibrary2-167.ino.elf > /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/TFTGraphicsLibrary2-167.ino.map"
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-size -A /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_397305/TFTGraphicsLibrary2-167.ino.elf
Sketch uses 1300 bytes (7%) of program storage space. Maximum is 16384 bytes.
Global variables use 87 bytes (16%) of dynamic memory, leaving 425 bytes for local variables. Maximum is 512 bytes.
````

SpenceKonde, 2023-06-22:
Yeah  - hypothesis confirmed. I've pushed fixes for all those warnings - every one of them was a legitmate defect in the avrdude.conf.

The issue with it not recognizing the tinyisp is because they changed that in the new avrdude and I need to adjuist programmers.txt to accounr for it, and find the thread where this headache-inducing subject of the various subspecies of usbtinyisp was discussed

technoblogy, 2023-06-22:
Sorry to add to your workload!

technoblogy, 2023-06-22:
By the way - I generated those listings without a USBtinyISP plugged in (I left it at home) so that probably explains the "unable to open programmer" error.

SpenceKonde, 2023-08-13:
Oh....

So we don't actually KNOW tjhat there's a problem with the tinyISP ?

SpenceKonde, 2023-08-13:
Try latest, I checked in a bunch of PWM adjustments, and added to part specific documentation tables. 

No resolution to any avrdude 7.1 problems, we are skipping 7.2 and doing 7.2 - when I can fucking get binaries.