SpenceKonde/ATTinyCore #779: Suggested change to the documentation for PWM on ATtiny87/167

technoblogy, 2023-06-21:
I feel that the paragraph about PWM on the ATtiny87/167 is confusing because it talks in terms of implementation, ie duty cycle, rather than function, which is generating an analogue voltage with `analogWrite()`:

https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/avr/extras/ATtiny_x7.md#flexible-pwm-support

Here's a suggested alternative:

> Timer1 has two channels, A and B, allowing you to generate two independent voltages with `analogWrite()`. One of them can be output on PB0, PB2, PB4, and/or PB6, and the other can be output on PB1, PB3, PB5, and/or PB7. Changing the voltage on one of these pins will affect all the other pins used with `analogWrite()` in the same group. To stop the PWM output call `digitalWrite()`, or `analogWrite()` with 0 or 255, on the pin.

SpenceKonde, 2023-06-21:
But, there's no analog voltage being generated anywhere.  I am not going to perpetuate the fairytales of the official documentation if they inaccurately describe PWM as an analog voltage.. Because it's NOT, and it doesn't act like an analog voltage would either. The only thing analog about analogWrite() (except on modern AVRs with a DAC, on that one pin) is the name. 

Throughout any of my cores, if any of my documentation anywhere refers to PWM output as a voltage, that is a defect in the documentation and should be reported as such, as it does not accurately describe what the hardware does

An analog voltage and a PWM duty cycle are not interchangeable, and acting as if they are the same thing does nobody any good

technoblogy, 2023-06-21:
OK, I see what you mean. But I still think it could be worded a bit more clearly. Here's a second attempt:

> Timer1 has two channels, A and B, allowing you to generate two independent duty cycles with `analogWrite()`. One of them can be output on PB0, PB2, PB4, and/or PB6, and the other can be output on PB1, PB3, PB5, and/or PB7. Changing the duty cycle on one of these pins will affect all the other pins used with analogWrite() in the same group. To stop the PWM output call `digitalWrite()`, or `analogWrite()` with 0 or 255, on the pin.

SpenceKonde, 2023-06-21:
Sure that works, I can copy paste it in or your can PR it for the glory.

technoblogy, 2023-06-22:
Please copy paste it. I'm not worried about glory!

technoblogy, 2023-06-23:
Changed my mind - I'll PR it (together with some other minor comments).

closed 2023-09-09 by SpenceKonde