SpenceKonde/ATTinyCore #787: No serial on 2

ahshah, 2023-07-06:


ahshah, 2023-07-06:
Codeline is currently on d670622155a459bf6e0757ac2032d0495bfa71b0: Correct issues with avrdude,conf.m make wiring readable

ahshah, 2023-07-06:
The power and gnd rails read 3.3 v , and you can see 3.3v is soldered in the on configuration

SpenceKonde, 2023-07-29:
Curious indeed. There isn't much to go wrong in serial on 841 - but some changes were made so it is possible that something was broken, this is a very serious bug seeing as the whole reason this core was started was to get a tiny with 2 serial ports. 

Can you post the .lst and .map files that are created when you export compiled binary? I know it does listings, and I think maps are in too, the map is actually what I'm more interested in for once...

ahshah, 2023-08-10:
Hi please find the requested resources attached. Codeline remaints at https://github.com/SpenceKonde/ATTinyCore/commit/d670622155a459bf6e0757ac2032d0495bfa71b0
[Serial.Attiny841.lst.map.zip](https://github.com/SpenceKonde/ATTinyCore/files/12316584/Serial.Attiny841.lst.map.zip)

SpenceKonde, 2023-08-13:
I may have fixed this issue ( am in middle of a move, don't have access to hardware to test on atm)

ahshah, 2023-08-14:
Confirmed, thanks for the fix! Confirmed working with 747d8533b747758be3bc4e91eb5c35a05057467c.
Could you narrow down what in the last few changes addressed this issue? Just for my edification.

SpenceKonde, 2023-08-14:
It was that big commit with all those updates to the part-specific docs. That one also has a change to HardwareSerial (maybe several of the hardware serial files). It was failing to retain the register addresses. They were passed to the constructor and flushed down the toilet.

ahshah, 2024-01-06:
Hi, me again.
Seems like I'm getting corruption on the serial when running at slower speeds. Similar sketch, different result:

```
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop() {
  Serial.println("on");
  delay(1000);  // wait for a second
  Serial.println("off");
  delay(1000);  // wait for a second
}
```

Results in gibberish-y output like so:
```
offonofCo                                     offon
         ofC
oofCoofCo   ofCon
oVH      HonoffConoff
   oHooffConCoffC
oHoCHonCofCoofCoC5VHonofCoCofVon
                  o
oHo                ofVon
   5
ofCooofCon
oVHoCofVHon5
            onC
```
Serial settings is 9600 @ 8N1
I've tested and reproduced this corruption on the following two commit points, but not inbetween.

`747d853`
`5508f88` 

Board settings as follows:
[image]

ahshah, 2024-01-08:
Hi it looks like this issue with baudrate has to do with internal oscillator variance, as similarly discussed here: 
https://forum.arduino.cc/t/attiny-841-serial-comms-baud-rate/343701

My attiny 841 is running off 5v so it seems that this is the likely culprit