SpenceKonde/ATTinyCore #791: digitalPinToInterrupt on ATTiny841

3sigma, 2023-07-13:
Hello,

On the following line you associate INT0 with PB2:

https://github.com/SpenceKonde/ATTinyCore/blob/e5046a5ab4cd68e7ca5e1906faa145f821d7358d/avr/variants/tinyx41_cw/pins_arduino.h#L94

But on the ATTiny841, INT0 is on PB1. Did I miss something ?

Thanks,

Nicolas

SpenceKonde, 2023-07-13:
And just a few days agpo someone asked why this version wasn't in board manager. It's stuff like this!!! Was wrong on both directions of pin mapping for the the x41, because the files had been based on the 84, and because Atmel's designers had been sniffing glue or something, the INT0 pin which was PB2 on the 84, was moved to the strictly worse PB1 for the follow-on product.

closed 2023-07-13 by SpenceKonde

3sigma, 2023-07-13:
Thank you. Yes, strange to use one of the crystal pin for this interrupt...

SpenceKonde, 2023-07-13:
Especially when the chip is a followon tp a chip that had it 9n a different
pin. M9re than anything I think what doomed Atmel to being bought out was a
lack of managers who were willing to walk the engineering floor wielding
whips to keep the engineers in line, and keep that darned dartboard that
they taped product package diagrams to in order to choose pinouts locked
up.
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***