SpenceKonde/ATTinyCore #800: http://drazzy.com/package_drazzy.com_index.json is not working anymore for Github actions

ArminJo, 2023-08-31:
Hi,
since a few days the URL http://drazzy.com/package_drazzy.com_index.json
is not working anymore, see https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/6041049256/job/16393168027#step:3:70.
Here, https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/5931549426/job/16083624590#step:3:70 it worked one week before.

Maybe Github force **https** now and has problems with the self signed certificate, as I have in Firefox with https://drazzy.com/package_drazzy.com_index.json.

Thanks for your work and for caring.
Best regards
Armin

SpenceKonde, 2023-08-31:
No, the certificate expired like it does about half the time it runs out because let's encrypt which seems to be the only way to get https certs gives certs that last such a short time that the proper tool to time them with is a stopwatch. I can't count how many times I've followed their autorenew setup instructions and shit don't ever autorenew. 

But I just got back after spending the past two weeks moving house - which means also moving my business and all my equipment, and it's just to a temporary place until I can find permanent accommodations because I lost the annual housing scramble (first time I've *had* a scramble to find a place - before I was always with a group, but my group split ways. I had people to renew at the end, but by then the place was rented to someone else. Moving costs = close to 2 months of rent, not counting the damage to my stuff. So far, just on the moving out phase (I'm sure there's more damage to be found), I'm down 2 sets of shelves beyond repair, and need to fix my desk lamp. Oh and the set of drawers I keep my components in fell off the cart when the movers took a corner too fast in the storage place, and lost a wheel, which I need to find a replacement for, or find new drawers, but I'm already down an additional pair of drawers... and I need to put new plugs on the hot air gun and one of the vacuum tweezers...) 

Everyone was all "you need to use https you need to use https" 

Well this is what https looks like - and unless someone knows a place that will give certs without charging an arm and a leg, or how make autorenew for letsencrypt work on amazon linux 2023, this is probably going to happen every 2 months, until azduino.com mkII is set up and at that point it'll happen every month because they both have to have certs and be https always cause people get their shorts in a knot when you serve anything over http.

SpenceKonde, 2023-09-02:
I can find not a trace that I ever had a working cert there. None of the tools are installed, nothing. How has this gone unnoticed?

I suspect that the answer is that they were using http://drazzy.com/package_drazzy.com_index.json 

This URL works. 

Despite the title of this issue, your request appears to be made against 
https://drazzy.com/package_drazzy.com_index.json 

This URL has an invalid self-signed certificate **and has never worked**. I had intended to set it up, and had apparently hallucinated that I did so (I must have, as there's no trace of anything related to setting up the cert, I have no notes, and despite it's widespread usage (AL2023 is the current "chosen linux" of the largest cloud computing provider and let's encrypt is huge), it seems that making them play well is not as easy as it was on the previous version of amazon linux (presumed collateral damage? Amazon isn't at war with Let's Encrypt - and in fact seems to carefully avoid certificate issuance. 

So I conclude that **https never worked** on drazzy.com; thus **this does not represent a regression, and is instead an enhancement request** and it will be addressed in due course, in the meantime, use a normal http request, not an https request. Https does not work correctly on drazzy.com at this time.. 

Prior to this summer, it didn't even open port 443. That system eventually failed early this summer, which was catastrophic as it blocked installation of the cores. Logs showed it dying mid-transfer, and after reboot it didn't return. Upon login, investigation found that the portion of the filesystem that had contained wwwroot no longer was present, and the containing directory was listed as if it were a file. 

That instance was flushed and a new one deployed (it was overdue for update for financial reasons too, it was on the cheapest instance when I'd first created it, but now they've introduced significantly cheaper ones). At that time I had intended to finish setup of https. Apparently I had gotten to the point where I was confounded by all these different sets of instructions for installing certbot on AL2023, all of them *completely different*, and not gone further except in my head.

Summary:
1. The error is caused by using a URL which was never valid. Use http[ 
2. The fact that it was intended to be valid is immaterial, as the wrong URL was not promoted yet.
3. It is not a regression, this is a feature enhancement. Accordingly, I don't see why this should be a particular priority. I'm going to move this issue a more approprate location (with all the other board manager packaging stuff) as this is by no means specific to this core.
4. Your tests seem to be able to falsely report pass resu