SpenceKonde/ATTinyCore #802: 'ATTinyCore by Spence Konde' doesn't show in Board Manager list

Napoleon-BlownApart, 2023-09-03:
Using Arduino IDE 2.2.1 on Mac, adding the `http://drazzy.com/package_drazzy.com_index.json` url to the list (only one in my case) and pressing OK, causes the IDE to download the `package_drazzy.json` file into `/Users/<user>/Library/Arduino15`.  But when I go into the Board Manager and try to find `ATTinyCore by Spence Konde` in the list, it's not there.

What to do?

Napoleon-BlownApart, 2023-09-03:
I installed Arduino IDE 2.1.0, and now it's working again.  (Didn't work with v2.2.0)

SpenceKonde, 2023-09-03:
Do note that we do not yet officially support IDE versions 2.x - as some people yourself included have observed, under at least some conditions, some 2.x.x versions work. I have not characterized which ones do or not (I'm sorry, I'm sick and tired of 80-90% of versions that Arduino ships being unfit for purpose with major regressions, and people acting like it's my job to figure that out). I asked way back when 2.0 first came out if anyone had tried it out or found any problems, on all of my cores. You could hear the chirping of crickets. But then I see issues getting created like this, with peiople 

Did any errors come up when you opened board manager? 

There is an issue with the webserver configuration; if the new IDE is trying to force https when http urls are fed to it, that would prevent it from downloading the board manager json file, because the https functionality of the webserver is busted in ways that I do not understand.

Napoleon-BlownApart, 2023-09-03:
I understand.  I believe this to be an issue with the IDE, not your package.
Under 2.2.1, no errors were displayed, just the entry was not visible in the board manager list after adding it to the preferences list.
Under 2.1.0, ATTiny85 and ATTiny88 (MH-ET Live) work fine for me using the Micronucleus.

redactedaccount, 2023-09-08:
I can't even get the `http://drazzy.com/package_drazzy.com_index.json` link to work, says it wont connect.

Napoleon-BlownApart, 2023-09-08:
> I can't even get the `http://drazzy.com/package_drazzy.com_index.json` link to work, says it wont connect.

Maybe try it in your browser?  Note it's not using `https`.
You could try, after doing the normal steps in the preference settings, manually download the file and save it into your `!/Library/Arduino15` folder using the name `package_drazzy.com_index.json`.
And then try to install `ATTinyCore` the normal way.

redactedaccount, 2023-09-08:
>Maybe try it in your browser?

Did YOU try it in your browser?

SpenceKonde, 2023-09-08:
Yes it is down because I'm trying to fix 5hw last problem reported but
broke my wrbsrtveterer in the procwss

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

redactedaccount, 2023-09-09:
Still down

SpenceKonde, 2023-09-09:
Still aware of the problem

skyreacher, 2023-10-06:
> Maybe try it in your browser? Note it's not using `https`. You could try, after doing the normal steps in the preference settings, manually download the file and save it into your [...]

I also ran into the issue, that the boards manager URL is not working (on Mac with Arduino IDE 2.2.1). Downloading the json file and putting it into ~/Library/Arduino15 was an easy work around. Thanks for sharing!

bjfischb, 2023-10-07:
I have the same issue as the original post: I can put the json in my preferences, and I then see the json file appear in my Arduino15 directory. However, ATtinyCore does not show up in the board manager as an option to download. I have since successfully downloaded a different board package, so it must be something about ATtinyCore and this release of the IDE. I'm using Arduino IDE 2.2.1, so I guess there's something that wasn't fully tested.

I'm able to do the manual download where I put the ATtinyCore package in my sketches hardware directory. But, it would be good to get this resolved so that the core can be maintained through the board manager. 

Is there any indication on what the root cause is? Does it have something to do with how the json file is named?

The ATtinyCore is working for me on IDE 2.2.1 as far as I can tell......

per1234, 2023-10-08:
Hi all. This is caused by a bug in Arduino CLI versions 0.34.0 - 0.34.2, which also affects Arduino IDE versions 2.2.0 - 2.2.1:

https://github.com/arduino/arduino-cli/issues/2345

I apologize for any inconvenience this has caused.

---

The bug has already been fixed, but there hasn't been a release of Arduino CLI since that time so the fix is not yet available to Arduino IDE users.

I'll provide instructions for a workaround:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Open the following folder in your file manager program (e.g., [Windows **File Explorer**](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1)):
   - **If you are using Linux:**
     ```text
     /home/<username>/.arduino15/
     ```
     (Where `<username>` is your Linux username)
     ❗ The `.arduino15` folder may be hidden by default in your file manager and terminal.
   - **If you are using macOS:**
     ```text
     /Users/<username>/Library/Arduino15/
     ```
     (Where `<username>` is your macOS username)
     ❗ The `Library` folder is hidden by default. You can make it visible by pressing the <kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**.**</kbd> keyboard shortcut.
  - **If you are using Windows:**
     ```text
     C:\Users\<username>\AppData\Local\Arduino15\
     ```
     (Where `<username>` is your Windows username)
     ❗ If looking for it with your file manager or command line, note that the `` folder is hidden by default. On [Windows "**File Explorer**"](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1), you can make it visible by opening the "**View**" menu, then checking the box next to "**☐ Hidden items**".
1. You will see a file named `package_drazzy.json` in that folder. Rename it to `package_drazzy.com_index.json`.

Now start Arduino IDE again. You should now see **ATTinyCore** (as well as the other SpenceKonde/DrAzzy) platforms listed in Boards Manager and be able to install it as expected.

SpenceKonde, 2023-10-08:
<3

closed 2023-10-08 by SpenceKonde

MyVanitar, 2023-11-19:
I followed your procedure for windows but I received the following error: (IDE: 2.2.1)

`
Failed to install platform: 'ATTinyCore:1.5.2'. 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip": tls: failed to verify certificate: x509: certificate has expired or is not yet valid:
`

SpenceKonde, 2023-11-19:
Yes, I bloody know this and I'm not allowed to do anything else until I fix
it, except for time bounded things like "ship this because the customer is
annoyed that it hasn't shipped yet" and "get your things packed and in the
car and get in because it's leaving in 2 hours, and you have no other way
to get out of here"

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

MyVanitar, 2023-11-19:
My intention was just to report this. otherwise, you are not bound to fix or reply. We haven't purchased anything from you to ask for aftersales or warranty. I installed this manually and it works flawlessly

per1234, 2023-11-20:
> ```text
> Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip": tls: failed to verify certificate: x509: certificate has expired or is not yet valid
> ```

That error is completely unrelated to the bug in Arduino CLI's handling of the package index filename (https://github.com/arduino/arduino-cli/issues/2345) reported here. There is a dedicated issue tracking the installation failure caused by the expired SSL certificate on azduino.com at https://github.com/SpenceKonde/ATTinyCore/issues/815. I know you already found a workaround **@

smdg49, 2023-11-22:
Between both workarounds I got it to work.
@SpenceKonde : thanks so much for ATTinyCore. 
Same as @MyVanitar, was just reporting the issue. Great work. You do not owe anyone anything.

Intechgreater, 2024-01-15:
> Hi all. This is caused by a bug in Arduino CLI versions 0.34.0 - 0.34.2, which also affects Arduino IDE versions 2.2.0 - 2.2.1:
> 
> [arduino/arduino-cli#2345](https://github.com/arduino/arduino-cli/issues/2345)
> 
> I apologize for any inconvenience this has caused.
> 
> The bug has already been fixed, but there hasn't been a release of Arduino CLI since that time so the fix is not yet available to Arduino IDE users.
> 
> I'll provide instructions for a workaround:
> 
> 1. Select **File > Quit** from the Arduino IDE menus if it is running.
> 2. Open the following folder in your file manager program (e.g., [Windows **File Explorer**](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1)):
>    
>    * **If you are using Linux:**
>      ```
>      /home/<username>/.arduino15/
>      ```
>      
>      
>          
>            
>          
>      
>            
>          
>      
>          
>        
>      (Where `<username>` is your Linux username)
>      ❗ The `.arduino15` folder may be hidden by default in your file manager and terminal.
>    * **If you are using macOS:**
>      ```
>      /Users/<username>/Library/Arduino15/
>      ```
>      
>      
>          
>            
>          
>      
>            
>          
>      
>          
>        
>      (Where `<username>` is your macOS username)
>      ❗ The `Library` folder is hidden by default. You can make it visible by pressing the **Command**+**Shift**+**.** keyboard shortcut.
> 
> * **If you are using Windows:**
>   ```
>   C:\Users\<username>\AppData\Local\Arduino15\
>   ```
>   
>   
>       
>         
>       
>   
>         
>       
>   
>       
>     
>   (Where `<username>` is your Windows username)
>   ❗ If looking for it with your file manager or command line, note that the `` folder is hidden by default. On [Windows "**File Explorer**"](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1), you can make it visible by opening the "**View**" menu, then checking the box next to "**☐ Hidden items**".
> 
> 1. You will see a file named `package_drazzy.json` in that folder. Rename it to `package_drazzy.com_index.json`.
> 
> Now start Arduino IDE again. You should now see **ATTinyCore** (as well as the other SpenceKonde/DrAzzy) platforms listed in Boards Manager and be able to install it as expected.

Greatly appreciate this!
How about adding it to the official installation instructions in the meantime? [https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/Installation.md](url)

SpenceKonde, 2024-01-15:
GOD THEY STILL HAVEN'T RELEASED AN UPDATE FOR THE IDE? What the hell is wrong with them?! They broke a major feature of the IDE in august. They had implemented the fix in september. FOUR MONTHS LATER WE ARE STILL WAITING FOR A RELEASE WITH THE FIX!

I do not make a policy of extensively documenting workarounds like this because the updates take so long to make it to the IDE that by then I've forgotten about all the places that I put such instructions, and then am mortifed a year or two later when someone asks "Is this being maintained? That workaround hasn't been needed for a year!" - so the problem is that I don't have a way to know where they all are, and I don't get notifications to know when to remove them, and the Arduino team frequently breaks something else when they fix something, so for a period of time the what would also be unclear. Since we have no good way to determine the where, no automatic way to know when, and no guarantee of what should be done to update the documentation, I just state that the 2.x.x versions have yet to include any version with basic functionality intact, and a fix is implemented but not available in any releases.

I also disapprove of extensively documenting such things because it minimizes the apparent severity and gives the Arduino team more of a free pass than I think they deserve. Especially considering the fact that it preferentially hits my cores because I'm apparently the only one who actually read their spec and named the board manager json file the way they said to - then they broke that, fixed it, but have been sitting on the fix for 4 months. 

Oh, and misdiagnosis of the problem and a botched attempt at correcting that issue is what led to this core being uninstallable for 3 months due to the expired SSL cert (the actual problem behind that was that I had been in the wrong window and deleted all of the apache configuration files from the live server that I was going to take down once the new one was set up instead of the new one, so while azduino.com's SSL cert worked, apache was still working from RAM, even though the certs and all the config files were deleted from the disk. but once the SSL cert expired, I couldn't renew it, and a bunch of shit came up such that a top priority task took 3 months to look at. 

I do not have time to document workarounds for this shit. I am struggling to keep my head above water as it is.

CaptainCheeze29, 2024-06-24:
I know you've got a lot on plates. Just updating. Boards were fine on 2.3.1 and now not fine on 2.3.2 again.

farukmath, 2025-01-04:
I had the same problem.
there is a solution in the video below.

https://www.youtube.com/watch?v=sycSdI49hlY