SpenceKonde/ATTinyCore #815: Fails to install

Alt0S4x, 2023-10-28:
Hello, when I try to install I get the following error:

Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Downloading packages
arduino:avrdude@6.3.0-arduino18
ATTinyCore:micronucleus@2.5-azd1b
Failed to install platform: 'ATTinyCore:avr:1.5.2'.
Error: 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: 

Thanks!

Juanchumu, 2023-10-28:
Hellooo, im having the same problem :/  

i dont know anything about certificates expires on pages, im gona wait

Error: 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: current time 2023-10-28T01:19:22-03:00 is after 2023-10-26T15:15:54Z

SpenceKonde, 2023-10-28:
Quoting the previous messages i've sent via email.
```
26 okt. 2023 kl. 22:29 skrev Spence Konde <[spencekonde@gmail.com](mailto:spencekonde@gmail.com)>:

Ugh. No ETA on fix because that server is trashed and the webserver is not belived to be restartable. We're in ther middle of a server migration that I thought was going to be done before the cert ran out. Until that migration is complete [azduino.com](http://azduino.com/) will not be able to service https requests. I apologize for any inconvenience.

This is a critical priority issue for us, however, there are 3 even more critical priority issues that must be resolved even more urgently.
```

1/3 more important tasks have been completed.

per1234, 2023-10-28:
Instructions for a workaround to the problem are provided here:

https://forum.arduino.cc/t/2-2-1-board-manager-download-additional-boards/1182930/28

Alt0S4x, 2023-10-28:
Thanks for the quick response! Spence, and thanks for the workaround perl1234!

santanig5, 2023-11-01:
Greetings,
I tried the workaround but I'm still getting an error.
IDE V1.8.19
placed the micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip file in the AppData\Local\Arduino15\staging\packages directory.
Also tried the manual unzip into the hardware folder I created but received IDE Error
Could not find boards.txt in C:\Users\User\Documents\Arduino\hardware\micronucleus-cli-2.5-azd1b-x86_64-mingw32\Drivers. Is it pre-1.5?

What else can I try?

kcl1s, 2023-11-03:
> Greetings, I tried the workaround but I'm still getting an error. IDE V1.8.19 placed the micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip file in the AppData\Local\Arduino15\staging\packages directory. [...]

My error was for a different file. I had to change the package download to ( micronucleus-cli-2.5-azd1b-i686-mingw32.zip ) for my win11 computer IDE Ver 1.8.19. That got the job done.

santanig5, 2023-11-03:
Thank you, unfortunately I still have the same issue.

[cid:26a8bc95-f6dd-4eae-929f-025ac6b4de70]

samster395, 2023-11-27:
Taken from https://forum.arduino.cc/t/2-2-1-board-manager-download-additional-boards/1182930/28

The workaround will be to manually download the tool archive (the archive download URLs can be extracted from [the package index file 12](https://drazzy.com/package_drazzy.com_index.json)) and then place it in the Boards Manager download cache. Arduino IDE checks for the file in the cache before attempting to download it from the Internet, so if you have manually placed the file in that location, you avoid the problematic download that causes the ATTinyCore installation to fail.

1. Download the ATTinyCore:micronucleus@2.5-azd1b Archive File
Unfortunately you will likely find that attempting to download this file via the browser is not very user friendly. For security reasons, the browser will try to strongly discourage from downloading it and you will have to make some efforts to overcome its resistance.

This is the download URL:

If you are using Windows:
http://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip
If you are using Linux x86:
http://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2
If you are using Linux ARM:
http://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-arm-linux-gnueabihf.tar.bz2
If you are using macOS:
http://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-apple-darwin.tar.bz2
2. Add the Downloaded File to the Boards Manager Cache
After you manage to download the file, copy it to the following folder:

If you are using Windows:
C:\Users\<username>\AppData\Local\Arduino15\staging\packages\
(Where <username> is your Windows username)

If you are using Linux:
/home/<username>/.arduino15/staging/packages/
(Where <username> is your Linux username)

If you are using macOS:
/Users/<username>/Library/Arduino15/staging/packages/
(Where <username> is your macOS username)

3. Install ATTinyCore
Now try installing ATTinyCore via the Arduino IDE Boards Manager again. The installation should now be successful.

per1234, 2023-11-28:
@samster395 the paths in step (2) were corrupted because you didn't put them in [code blocks](https://docs.github.com/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) as they were on the Arduino Forum post. I fear this will cause confusion for the people trying to follow the instructions. Please either fix the formatting of the instructions or else delete your posts.

samster395, 2023-11-28:
> @samster395 the paths in step (2) were corrupted because you didn't put them in [code [...]

I have fixed it, good call.

ustajan, 2023-12-09:
This is to confirm that the .json "standard" path still doesn't work because of certificate issues, **however** the workaround describe [above](https://github.com/SpenceKonde/ATTinyCore/issues/815#issuecomment-1828515013) (also [here](https://forum.arduino.cc/t/2-2-1-board-manager-download-additional-boards/1182930/28)) works .  I was able to install ATTinyCore.

SpenceKonde, 2023-12-20:
I'm surprised you were able to install with any workaround, I must not have the servers in the state I thought they were - the old azduino shouldn't be running now, and the new one hasn't been set up...

SpenceKonde, 2024-01-08:
This is fixed now.

closed 2024-01-08 by SpenceKonde

Einheit-101, 2024-11-10:
WHY in GODS NAME is this issue still a thing?? It CANT be that hard to just install AttinyCore for Arduino IDE, is it?

per1234, 2024-11-11:
The current occurrence of the website certificate problem is being tracked at https://github.com/SpenceKonde/ATTinyCore/issues/885. Check **@SpenceKonde**'s comments on that thread for an update on the situation.