SpenceKonde/ATTinyCore #816: arduino-cli: installation error

pstolarz, 2023-10-28:
Trying to install the core from `arduino-cli` causes error as follows:
```
root@f15b4e6e2442:/# arduino-cli config add board_manager.additional_urls http://drazzy.com/package_drazzy.com_index.json
root@f15b4e6e2442:/# arduino-cli update
Downloading index: library_index.tar.bz2 downloaded                                                                                                                                 
Downloading index: package_index.tar.bz2 downloaded                                                                                                                                 
Downloading index: package_drazzy.com_index.json downloaded                                                                                                                         
Error initializing instance: Loading index file: loading json index file /root/.arduino15/package_drazzy.com_index.json: open /root/.arduino15/package_drazzy.com_index.json: no such file or directory
```

As I see a root-cause of the problem is already known: https://github.com/arduino/arduino-cli/issues/2345

Maybe it's north to provide alternative name of the files like `package_drazzy_com_index.json` to make the installation possible?

pstolarz, 2023-10-30:
Seems the bug will be fixed in the next arduino-cli release by https://github.com/arduino/arduino-cli/commit/dc5c56eb1a1250c8d8e05514d0bdebb970ba2658. Closing the issue here.

closed 2023-10-30 by pstolarz

SpenceKonde, 2023-10-31:
Yeah but it still won't install because of the other outstanding issue