SpenceKonde/ATTinyCore #819: ATtinyCore installation failed due to expired tls certificate for azduino.com

schorsch3000, 2023-11-09:
Title says it all basically.
While installing ATtinyCore 1.5.2 in Arduino 2.2.1 (using the rename package_drazzy.json workaround) ill get the following error in the console:
```
Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Downloading packages
arduino:avrdude@6.3.0-arduino18
ATTinyCore:micronucleus@2.5-azd1b
Failed to install platform: 'ATTinyCore:avr:1.5.2'.
Error: 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: current time 2023-11-09T18:03:41+01:00 is after 2023-10-26T15:15:54Z

```

roger-tannous, 2023-11-09:
It seems that the SSL certificate (from Let's Ecrypt) is expired.

I have faced the same problem, and fixed it like the below (as a temporary arrangement)::

On Linux/Mac, 

sed -ri "s/https\:\/\/([^\"]+)/http\:\/\/\1/g" ~/.arduino15/package_drazzy.com_index.json

sed -ri "s/https\:\/\/([^\"]+)/http\:\/\/\1/g" ~/.arduino15/package_drazzy.json

On Windows, just edit the file(s) and change https to http for all URLs that are double quoted.

roger-tannous, 2023-11-09:
It seems this issue has been handled in a different way previously, here's [the link](https://forum.arduino.cc/t/2-2-1-board-manager-download-additional-boards/1182930/2)

per1234, 2023-11-17:
Duplicate of https://github.com/SpenceKonde/ATTinyCore/issues/815

closed 2023-12-20 by SpenceKonde

schorsch3000, 2023-12-21:
How is this completed?
Nothing changed...

hmeijdam, 2023-12-21:
> How is this completed? Nothing changed...

Because it's a duplicate report of an already existing issue. That's what the "duplicate" label means.

[image]

schorsch3000, 2023-12-21:
[image]

There are options :-)