SpenceKonde/ATTinyCore #820: ATTinyCore does not appear in Boards Manager

roger-tannous, 2023-11-09:
OS:: Ubuntu 22.04.3 LTS (Jammy Jellyfish)

Arduino IDE version 2.2.1

URL added to Additional Board Manager URLs:: http://drazzy.com/package_drazzy.com_index.json

But ATTinyCore does not appear in Boards Manager, even after multiple restarts of the Arduino IDE.

I don't know what's the optimal solution for this issue, but I've solved it this way::

Close the Arduino IDE and **relaunch it using the command line**.

Watch the logs (when launched from the command line, the application dumps the logs to the standard output)

2023-11-09T19:59:34.321Z root ERROR Detected an error response during the gRPC core client initialization: code: 9, message: Loading index file: loading json index file ~/.arduino15/package_drazzy.com_index.json: open ~/.arduino15/package_drazzy.com_index.json: no such file or directory

Let's find what's there

find ~/.arduino15/ -type f -name "package_drazzy*.json"

~/.arduino15/package_drazzy.json

check if the file has been messed with

wget http://drazzy.com/package_drazzy.com_index.json

diff -qs package_drazzy.com_index.json ~/.arduino15/package_drazzy.json

Files package_drazzy.com_index.json and ~/.arduino15/package_drazzy.json are identical

mv ~/.arduino15/package_drazzy.json ~/.arduino15/package_drazzy.com_index.json

restart Arduino IDE

Why does it create package_drazzy.json again ?!

find ~/.arduino15/ -type f -name "package_drazzy*.json"

~/.arduino15/package_drazzy.com_index.json
~/.arduino15/package_drazzy.json

Now that ~/.arduino15/package_drazzy.com_index.json is there, ATTinyCore appears in the Boards Manager.

santanig5, 2023-11-13:
I am having the same issue with Windows 10 using IDE 2.2.1.
Device name	Cooler-Master-Desktop
Processor	Intel(R) Core(TM) i5-7400 CPU @ 3.00GHz   3.00 GHz
Installed RAM	16.0 GB
Device ID	7CD9120B-DF56-4CD1-AC90-01E71825AC53
Product ID	00325-80060-72834-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 10 Home
Version	22H2
Installed on	‎9/‎12/‎2020
OS build	19045.3570
Experience	Windows Feature Experience Pack 1000.19052.1000.0

URL added to Additional Board Manager URLs:: http://drazzy.com/package_drazzy.com_index.json

But ATTinyCore does not appear in Boards Manager, even after multiple restarts of the Arduino IDE.

Same problem on all three of my Windows PCs.

Please help, I am unable to complete my project.

Thanks

smdg49, 2023-11-14:
I opened this issue last week.

SonnyCarson86, 2023-11-16:
im having the same issue, attinycore doesnt appear in board manager

roger-tannous, 2023-11-16:
On Windows, you can do this::

With http://drazzy.com/package_drazzy.com_index.json added to the Additional Board Manager URLs,

rename C:\Users\%USERNAME%\AppData\Local\Arduino15\package_drazzy.json to C:\Users\%USERNAME%\AppData\Local\Arduino15\package_drazzy.com_index.json

Note that a copy won't solve the problem, you have to rename the file.

%USERNAME% is a Windows global variable that will automatically replace your username on your local machine.

Then, just because there's currently a problem with the SSL certificate of the domain from which micronucleus will be fetched, 
we can do [something similar to what's being mentioned here...](https://github.com/SpenceKonde/ATTinyCore/issues/819#issuecomment-1804659875)
which resolves to opening C:\Users\%USERNAME%\AppData\Local\Arduino15\package_drazzy.com_index.json in your favorite editor, and replacing all occurrences of **https** with **http** (CTRL+H then find: https replace with: http and hit the Replace All button).

Hope this helps!
Roger

santanig5, 2023-11-17:
Hello,
I followed the instructions, and it still does not work.
IDE 1.8.19 is still searching for https
[cid:25baf232-5560-48f0-9b35-b50137b3040c]

IDE 2.2.1 still has not ATTinyCore option and package_drazzy.json must be auto generated as I deleted it and when I restarted IDE it was back, I did edit both json files replacing the https with http, it replaced 212 entries.
[cid:1f5c9b04-f575-4a5f-8011-1f033b4ff2cf]
any other suggestions?

thanks,
Gary

ExIfDev, 2023-12-10:
hello, here is a fix to invalid certificate error when installing the package:

first you want to follow the steps indicated from [roger-tannous](https://github.com/SpenceKonde/ATTinyCore/issues/820#issuecomment-1814798675) then

you have to edit both files in:

C:\Users\%username%\AppData\Local\Arduino15

**package_drazzy.com_index.json**
and
**package_drazzy.json**

open them in notepad and **search** for this link: https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip

and **replace it** with this link (its hosted on my server and doesnt have certificate errors):

**http://aexadev.ddns.net/uploads/micronucleus-cli-2.5-azd1b-x86_64-mingw32.zip**

And it should work!

SpenceKonde, 2023-12-20:
Closing as duplicate; note that using an old version of the IDE won't actually let you install it either, because my webserver is hosed and needs to be rebuilt, among like a million other critical tasks

closed 2023-12-20 by SpenceKonde