SpenceKonde/ATTinyCore #836: ATTinyCore board manager JSON not working with Arduino IDE 2.2.1

uski, 2024-01-14:
Hi!

I have used the library before. As usual I added the JSON URL to the "Additional board manager URLs" section of the Arduino preferences screen, after a fresh install of the latest version of Arduino IDE 2.2.1.

Right after doing that, I see a popup at the bottom right of the screen saying it is indeed downloading the JSON file.

But to my surprise, after doing that, ATTinyCore does not show up in the available board managers!

The board manager URL I used is http://drazzy.com/package_drazzy.com_index.json

To validate that the problem comes from ATTinyCore, I tried adding the ESP8266 Arduino core (JSON https://arduino.esp8266.com/stable/package_esp8266com_index.json) and it worked just fine.

So, something is broken in the JSON of ATTinyCore, I have no idea how to debug this - let me know if I can help.

Thanks

uski, 2024-01-14:
I launched the Arduino IDE from a command line and watched the stdout.

`2024-01-14T03:57:42.899Z daemon INFO time="2024-01-14T04:57:42+01:00" level=info msg="Updating index" url="http://drazzy.com/package_drazzy.com_index.json"
2024-01-14T03:57:42.901Z daemon INFO time="2024-01-14T04:57:42+01:00" level=info msg="Starting download" url="http://drazzy.com/package_drazzy.com_index.json"
2024-01-14T03:57:42.903Z root INFO core-client-provider [platform-index] Downloading index: package_drazzy.com_index.json

2024-01-14T03:57:44.655Z daemon INFO time="2024-01-14T04:57:44+01:00" level=info msg="Checking signature" index="c:\\Users\\x\\AppData\\Local\\Arduino15\\package_index.json" signatureFile="c:\\Users\\x\\AppData\\Local\\Arduino15\\package_index.json.sig" trusted=true
2024-01-14T03:57:44.657Z root ERROR Detected an error response during the gRPC core client initialization: code: 9, message: Loading index file: loading json index file c:\Users\x\AppData\Local\Arduino15\package_drazzy.com_index.json: open c:\Users\x\AppData\Local\Arduino15\package_drazzy.com_index.json: The specified file cannot be found.`

I looked into `c:\Users\x\AppData\Local\Arduino15\` and interestingly, I can see a file named `package_drazzy.json` but not `package_drazzy.com_index.json`. This is a mismatch from the filename in the URL.

I do see a file named `package_esp8266com_index.json` which matches the ESP8266 board URL filename.

I did an experiment and copied the `package_drazzy.com_index.json` file onto my own web server, but I renamed it as `package_drazzycom_index.json` removing the dot like ESP8266 does.

**And this fixes the issue! So there is a problem with the filename of the library!**

To be 100% sure, I renamed `package_drazzycom_index.json` to `package_drazzy.com_index.json` (with the dot) on my webserver, and it stops working just like the official URL.

That's a problem... not sure what behavior is correct. I notice that the ESP32 arduino board URL is `https://espressif.github.io/arduino-esp32/package_esp32_index.json` and it does not have multiple dots. So I presume that multiple dots in the URL is the issue.

If so, I think a suitable solution would be:
- First, contact Arduino and verify if it is not a bug with IDE 2.2.1. If it is and they accept to fix it, just wait for them to do it and post a notice on the project page for users to use another version.
- If it is not a bug of the Arduino IDE, for instance if they specify to not use multiple dots, then you should start telling users to use a new URL without the dot like I did above, and you will have to keep both copies active forever to maintain compatibility.

Hope this helps

Intechgreater, 2024-01-14:
Having the same issue

hmeijdam, 2024-01-14:
Read all about it.
https://github.com/SpenceKonde/ATTinyCore/issues/802#issuecomment-1751882383

SpenceKonde, 2024-01-15:
Duplicate documented in 802. This is NOT OUR BUG, Please direct to the Arduino team any complaints about the fact that the fix was done 3+ months ago and still isn't in a release.

closed 2024-01-15 by SpenceKonde