SpenceKonde/ATTinyCore #847: SSL for site drazzy.com is expired

spellbinder103, 2024-02-22:
has troublesome when Arduino IDE download package_drazzy.com_index.json from site drazzy.com, is prompt site certificate expired.

AIMSTEC24, 2024-02-23:
Error downloading http://drazzy.com/package_drazzy.com_index.json
SAME PROBLEM, any one here ??

hmeijdam, 2024-02-23:
Please check this https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469

per1234, 2024-05-29:
This was resolved at one point, as was indicated by the closure of https://github.com/SpenceKonde/megaTinyCore/issues/1059 on 2024-03-29.

Unfortunately the renewed certificate expired once again yesterday, so we are back to the same problem as originally prompted the opening of this issue. Since this issue was never closed, I guess it will be best to reuse it for the new incarnation of the problem rather than opening a new issue for it.

viniciussilverio, 2024-06-24:
Hi all, for the ones who need a workaround, you can use a proxy to bypass the certificate.

I used this one: https://us10.proxysite.com/process.php?d=fKDkvVZ0eVvjm7pEdH2n6CvjweZNeQoK8NhAdsRuzbYnTjEKE19bqBa12yY%3D&b=1

You dont need to relly on the link I shared, you can use any proxy, this is just an example, but it also works

Hope it helps

SpenceKonde, 2024-09-08:
Corrected sometime ago.

closed 2024-09-08 by SpenceKonde