SpenceKonde/ATTinyCore #861: Is aTTiny core broken?

terrypin999, 2024-05-31:
I've just asked about this in the Arduin Forum, here:
[https://forum.arduino.cc/t/what-is-status-of-attinycore/1266483](url)
and copy my post below
--------------------
I'm encountering various inconsistent errors in my attempts to re-program my cheap Tiny 88 boards. I successfully did so in May 2023 and even posted a detailed step-by-step on how to do it In post #4 of this thread
[https://forum.arduino.cc/t/problem-loading-attinycore/1241472/4](url)
but I now fail to reprogram my Tiny 88 boards.

Researching, I'm confused about the status of Spencer Konde's core. Is it broken? The licence apears to have expired two days ago. And trying to open the URL
[http://drazzy.com/package_drazzy.com_index.json](url)

results in this:

[image]

In post #4 of this thread
[https://forum.arduino.cc/t/problem-loading-attinycore/1241472/4
](url)

@ptillisch said:

_"If we had a copy of the micronucleus-cli-2.5-azd1b-i686-mingw32.zip file, I could provide instructions for a workaround to the problem."_

Is that guidance now available please? FWIW I do have that file
C:\Users\terry\AppData\Local\Arduino15\staging\packages\micronucleus-cli-2.5-azd1b-i686-mingw32.zip

muffinista, 2024-06-01:
Looks like the certificate for drazzy.com is expired, which is preventing the arduino IDE from making downloads.

hmeijdam, 2024-06-01:
You can try using [this workaround ](https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469)

That worked fine last time when the certificate expired.

terrypin999, 2024-06-02:
Many thanks Hans, that fixed it for me.

SpenceKonde, 2024-09-08:
Certificates were restored upon my escape from two new peripherals which didn't like what I'd been saying bout em.

closed 2024-09-08 by SpenceKonde