SpenceKonde/ATTinyCore #862: Help with Attiny85 RX Uart Pin

GabrielSCorrea, 2024-06-01:
I may be getting , butI believe there is a discrepancy in the documentation regarding the RX pin for the ATtiny x5 family. The README.md states that the RX pin is on PA1, while the extras/avr documentation indicates it is on AIN1.

[image]
[image]
[image]

hmeijdam, 2024-06-01:
You are looking at the wrong part ATtinyx4 

[image]

backgroundprocessing, 2024-07-11:
For the ATtiny x5, the line should read "Software PB0, PB1" rather than "Software PB0, PA1".
There are no PA pins on the ATtiny x5 devices.

As shown on the pinout diagram, PB1 is also referred to as AIN1, either name can be used.

bakeddean, 2024-07-20:
I think @backgroundprocessing is correct, there is a typo in the documentation.
From https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/avr/cores/tiny/TinySoftwareSerial.h
```
/* Where should this work?
 * ATtiny x5:   Yes - RX on PB1, TX defaults to PB0, can be PB0~4.
 ```