SpenceKonde/ATTinyCore #869: Certificate expired for drazzy.com

ctnadovich, 2024-07-03:
It appears this expired Wed, 29 May 2024 01:49:13 GMT

Consequently, this package index URL http://drazzy.com/package_drazzy.com_index.json  can be problematic.

hmeijdam, 2024-07-03:
You can find the workaround when you read through the list of existing issues.

ctnadovich, 2024-07-03:
Then you should remove this broken URL (and the suggestion to use it) from the installation instructions.

hmeijdam, 2024-07-03:
I would, if I could, but the maintainer of this repository hes not been around for a while. I am just a user like yourself.

arekm, 2024-07-14:
https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/master/package_drazzy.com_index.json

VoidJuiceConcentrate, 2024-07-18:
Hello! Please fix the security certificate on your website drazzy.com. It appears to have expired on May 29th 2024. I believe you can use LetsEncrypt (https://letsencrypt.org/) for free short certs and cheap long term certs.

Alternatively, the .json file could be moved to github and you could operate out of there but I'm not sure how tenable that would be given the limits github imposes on data pulls.

VoidJuiceConcentrate, 2024-07-18:
> https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/master/package_drazzy.com_index.json (but still problematic as it points to some not existing archives)

I'm going to use this workaround for now, but it seems this is partially broken and definitely not what the official instructions say to use. (Also I'm not a big fan of using workarounds for security verification issues so I hope the cert thing gets fixed soon)

SpenceKonde, 2024-07-28:
fixed

closed 2024-07-28 by SpenceKonde