SpenceKonde/ATTinyCore #870: ATtiny core

Gi76t, 2024-07-17:
Hello Dear @SpenceKonde
I have an issue with downloading ATtiny core
Here's the error:
Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Downloading packages
arduino:avrdude@6.3.0-arduino18
ATTinyCore:micronucleus@2.5-azd1b
Failed to install platform: 'ATTinyCore:avr:1.5.2'.
Error: 2 UNKNOWN:

can you help?

hmeijdam, 2024-07-17:
As a workaround you can put this in your: File -> Preferences> Additional board manager URLs

https://archive.org/download/azduino_micronucleus-mirror/package_drazzy.com_index.json

There is a security certificate error now for a few months now. See other issues in the open issue list.

SpenceKonde, 2024-09-08:
This issue has been resolved.

closed 2024-09-08 by SpenceKonde