SpenceKonde/ATTinyCore #874: Attiny2313: error: 'PCIE' was not declared in this scope; error: 'PCIF' was not declared in this scope

badomenspeaking, 2024-08-24:
Arduino IDE 1.8.19. The following code gives the above mentione derror. Besedies them, also the following warnig:

"warning: 'PCINT_vect' appears to be a misspelled 'signal' handler, missing '__vector' prefix [-Wmisspelled-isr]"

#include <avr/io.h>
#include <avr/interrupt.h>

#define COUNTER_START_PIN        PB5  // MOM OFF MOM Rocker switch
#define COUNTER_STOP_PIN         PB4  // MOM OFF MOM Rocker switch
#define DISPLAY_RUN_PIN          PB3  // MOM OFF MOM Rocker switch
#define DISPLAY_FREEZE_PIN       PB2  // MOM OFF MOM Rocker switch
#define RESET_BUTTON_PIN         PB1  // MOM button
#define DISPLAY_PAUSE_SCK_PIN    PB0  // socket 9 of LH manual

volatile uint8_t inbyte = 0;

ISR(PCINT_vect) {
  if (!(PINB & (1 << COUNTER_START_PIN)))
    inbyte = 1;
  if (!(PINB & (1 << COUNTER_STOP_PIN)))
    inbyte = 2;
  if (!(PINB & (1 << DISPLAY_RUN_PIN)))
    inbyte = 3;
  if (!(PINB & (1 << DISPLAY_FREEZE_PIN)))
    inbyte = 4;
  if (!(PINB & (1 << RESET_BUTTON_PIN)))
    inbyte = 5;
  if ((PINB & (1 << DISPLAY_PAUSE_SCK_PIN)))
    inbyte = 6;

  Serial.write(inbyte);  
}

void setup() {
// Port B initialization
// Function: Bit7=In Bit6=In Bit5=In Bit4=In Bit3=In Bit2=In Bit1=In Bit0=In 
  DDRB=(0<<DDB7) | (0<<DDB6) | (0<<DDB5) | (0<<DDB4) | (0<<DDB3) | (0<<DDB2) | (0<<DDB1) | (0<<DDB0);
  // State: Bit7=T Bit6=T Bit5=T Bit4=P Bit3=P Bit2=P Bit1=P Bit0=P 
  PORTB=(0<<PORTB7) | (0<<PORTB6) | (0<<PORTB5) | (1<<PORTB4) | (1<<PORTB3) | (1<<PORTB2) | (1<<PORTB1) | (1<<PORTB0);

	cli();
	
  // External Interrupt(s) initialization
// INT0: Off
// INT1: Off
// Interrupt on any change on pins PCINT0-7: On
  GIMSK = (0<<INT1)   | (0<<INT0)   | (1<<PCIE);
  MCUCR = (0<<ISC11)  | (0<<ISC10)  | (0<<ISC01)  | (0<<ISC00);
  PCMSK = (0<<PCINT7) | (0<<PCINT6) | (1<<PCINT5) | (1<<PCINT4) | (1<<PCINT3) | (1<<PCINT2) | (1<<PCINT1) | (1<<PCINT0);
  EIFR  = (0<<INTF1)  | (0<<INTF0)  | (1<<PCIF);

  sei();
  
  Serial.begin(4800);
}  

void loop() {
}

hmeijdam, 2024-08-24:
Where did this code come from?

it uses register names that dont exist  e.g. "EIFR". there is a GIFR register though

for the GIMSK register there is no PCIE. You have to use PCIE1, PCIE2 or PCIE0 

[image]

The PCINT_vect vector does not exist 

you have PCINT0, PCINT1 and PCINT2 as available vectors

[image]

Read your datasheet

badomenspeaking, 2024-08-25:
Thanks for your reply.

Code was auto generated with CodeVisionAVR selecting the Attiny2313 uP, and
referenced to Atmel-2543-AVR-ATtiny2313_Datasheet.
After some trials and investigation, I discovered the actual reference is
Atmel - doc8246 - Attiny2313A, even though my uP is labelled
Attiny2313-20PU.
This solves the question. Thanks again.

Regards

Badomen

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Privo
di virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

hmeijdam, 2024-08-25:
The Attiny2313 was replaced by the Attiny2313A. I don't think the 2313 is supported in this core only its successor 

This [Atmel Application note](https://ww1.microchip.com/downloads/en/Appnotes/doc8261.pdf) describes some of the changes.

SpenceKonde, 2024-09-07:
Oh goddamnit I wrote an essay length response to this

Basically, there's some reason why I couldn't just have them be different options, some step in the path couldn't tell them apart (maybe avrdude?)... And the parts are very nearly identical. The 2313A is, in fact, fully binary compatible with 2313 code. However, said code will not compile for the 2313A - only the compiled output is compatible - this is because they changed the name of a small number of bitfields. I thought I was catching these and automatically pointing them to the new names, but maybe not. PCIE and PCIF were changed to PCIE2 and PCIF2, with the addition of PCINT support on the rest of the pins. 
Very very few differences overall. They can all be spotted by comparing the register map for 2313  and for 2313A - the A got some "reserved" bitfields made available and added a few new registers, and renamed a few others since they now needed numbers to differentiate them from their new friends. 

But yeah - since:
1) 2313 are vanishingly rare in the field, vastly outnumbered by 2313A's (the 2313 has been in "legacy surcharge" pricing since well before the microchip buyout, being around 50% more expensive. Combined with the fact that the 2313A is strictly better than the 2313, it's hardly surprising that the original version is now uncommon. 
2) AVRdude can't tell the difference between A and not-A.
3) Parts are so close to identical that as long as you adapt the names of the registers in the very few places they are different, and don't attempt to use any features that don't exist on the 2313, the result will work.

I decided that was acceptable.

badomenspeaking, 2024-09-07:
I agree.

closed 2024-09-08 by SpenceKonde