SpenceKonde/ATTinyCore #885: https://drazzy.com/package_drazzy.com_index.json - SSL-certificate expired 3 days ago

realvinx, 2024-10-24:
https://drazzy.com/package_drazzy.com_index.json - SSL-certificate expired 3 days ago

Please renew your certifiacte, Arduino IDE is complaining and I don't know how to skip the SSL check in Arduino IDE.

Jaepe, 2024-10-25:
You can install manually. [install.md](https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/Installation.md)
In your sketchbook location make a "hardware" folder if there isn't one already, then extract the zip from the repo into that folder. For me it looked like this "~/Arduino/hardware/ATTinyCore-1.5.2"

Jaepe, 2024-10-25:
Same as [Issue 869](https://github.com/SpenceKonde/ATTinyCore/issues/869)

ArminJo, 2024-10-30:
@SpenceKonde 
Hi Spence, 
thanks for your support of the small AVR CPU's 🥇 
But why not simple host the json on Github like e.g. ESP32 and other do?
A one time change is much better than all the breaks in availability.

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json

LordNoodles, 2024-10-31:
I downloaded and extracted the zip. I now have a folder named ATTinyCore-1.5.2 and within it a folder of the same name. 

which of the two do I copy to Documents/Arduino/hardware ?

LordNoodles, 2024-10-31:
thanks, i guessed correctly but you can't tell me you've never seen nested folders of the same name in some file structures. shit happens

SpenceKonde, 2024-11-01:
This issue is known. I am tryhing to locate the priviate key that I need to log into the machines, however this is challenging as the server they are backed up on is on the fritz (it's the fucking power connector, as alwatys. But the result is that only 20-40% of the time does it even complete the boot process before the power glitches and it turns off, and the laptop computer they were also located on failed not long ago. I then revived again while using a loaner computer, but I've serched EVERYEWHERE and can't find it. 
So as soon as I can locate the all important certificate (if I cannot, this project maty have to be shut down entirely becasuye I am unspeaklably sick of reconfiguring webservers tyop do the exact same tyhing

ArminJo, 2024-11-01:
So why not avoid half the trouble and at least put the json on Github?

SpenceKonde, 2024-11-01:
> So why not avoid half the trouble and at least put the json on Github?

I'd sooner kill myself than admit such sickening failure. Never suggest that to me again!

And besides I have no faith in those fuckers either. Every time IO've relied on an online service, year or two down the road, they yank it out from under my feet, or they secretly increase their monthly fees by an order of magnitude. I am still soo pissed at that registrar
Took me for THOUSANDS without my knowledge

brian-gee, 2024-11-05:
Manual install does not work either. Getting an "Error Fetching Resource".

lsellens, 2024-11-05:
> I'd sooner kill myself than admit such sickening failure. Never suggest that to me again!
>
> And besides I have no faith in those fuckers either. Every time IO've relied on an online service, year or [...]

You could always just mirror it on github, so if your server or you dies it will live on...

terrypin999, 2024-11-06:
> I downloaded and extracted the zip. I now have a folder named ATTinyCore-1.5.2 and within it a folder of the same name.
> 
> which of the two do I copy to Documents/Arduino/hardware ?

I'm confused about this too (as I usually am whenever I trouble-shoot issues with this core.) How/where did you get a recent folder called ATTinyCore-1.5.2? 

The lowest extracted subfolder from the latest ZIP download I've just done from
https://github.com/SpenceKonde/ATTinyCore
is named '**ATTinyCore-2.0.0-devThis-is-the-head-submit-PRs-against-this**' .  The folder's properties identifies it as version 2.0.0. Did you/others just rename it to the much shortger and more sensible name?

Adding to my confusion (probably down to me - not rare!) is that this folder has a size of  31,774,918 bytes, but an earlier version
(April) of exactly the same name is 63,545,738, suspiciouisly close to twice the size.

Given the license matter, can one of the experts also advise if I should still have this, not the original, in Preferences please?
https://descartes.net/package_drazzy.com_index.json

Terry, hobbyist end user

per1234, 2024-11-07:
We are supporting @terrypin999's request over on Arduino Forum:

https://forum.arduino.cc/t/confuision-again-re-spencekonde-attinycore-status/1319511

SpenceKonde, 2024-11-08:
I have bee. aware of this issue since it happ[ened
however until completing the next phase of work on my tax arrears ()not
inability to pay, inability to calculate how much to pay them, and irt
tookYRARS TO FIND A prepaRERE who qouylsn'r juqr say "Well give me X, Y amd
Z and I can get thyme prepared in a jiffy' "I don't know X Y oer Z" "Then
you nee3d to find thaT OUT, io CAN]T HELP YOU WITHOUT THAT. I DON'T EVEN
KNOW WHAT THOSE THINGS Are! i AM AN ENGINEER! i AM NOT AN ACCOUNTANT!
Finally I found a more helpful one, buty getttiering the data has bneen why
I've been so absent for a year.

The IRS is threatening to seize assets and charge me with tax evasio until
I get all ytjis sorted ouit so any time there is something to do on that, I
need to 100% instqantly dr0p all other activity and focu on that; I don't
want to go to jail and I don'ty want to have my worldly possessions lost.
When that jhappens befopre the cert that I jhave configured to
autorenw (burt which does't) happens to happening when I have a
Stay-Out-Of-Jail priority, ptohibity myelfg from doing anything else
otyherwise w2el;l, thaty's how I got 7 years behunbd.
,

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

jvasileff, 2024-11-08:
I think that's an uncharitable take, @gnif. No one is above splitting their attention with other responsibilities, vacation, time-off, etc. And the solution to service unavailability isn't to turn over hosting to some other single individual.

As others have mentioned, I think hosting on Github would be a great solution. Online services may fail or change policies, but less often than problems generally occur with individual servers. And, if additional project maintainers are brought on board, hosting on Github would allow others to help out when necessary.

kiwi64ajs, 2024-11-13:
Until the SSL issue is resolved, the Package JSON file can be accessed here: [https://web.archive.org/web/20240701151914/https://drazzy.com/package_drazzy.com_index.json](https://web.archive.org/web/20240701151914/https://drazzy.com/package_drazzy.com_index.json)

per1234, 2024-11-13:
@kiwi64ajs did you test it? That won't work for multiple reasons:

The first is that is the URL of the Wayback Machine web page for the file, not for the JSON file alone. Arduino IDE needs to download a JSON file it can parse, but if it downloads the page from that URL it will have a bunch of HTML in addition to the JSON content.

We can get around that by using [this feature of Wayback Machine](https://en.wikipedia.org/wiki/Help:Using_the_Wayback_Machine#Specific_archive_copy:~:text=if_%20or%20fw_%20Iframe%20%2D%20return%20document%20formatted%20normally%2C%20but%20without%20the%20navigational%20toolbar.):

> The following flags can be appended to the datetime field to modify the format in which the archived content is displayed
>
> [...]
>
> - `if_` or `fw_` Iframe - return document formatted normally, but without the navigational toolbar.

so the URL would be:

https://web.archive.org/web/20240701151914if_/https://drazzy.com/package_drazzy.com_index.json

(note the added `if_`)

Unfortunately we now run into yet another problem, which is that Wayback Machine has replaced all the original URLs in the file with the Wayback Machine equivalent URLs. For example, this line in the original file:

https://github.com/SpenceKonde/ReleaseScripts/blob/611e716c61335c1315cf83acff738b4d2eb67525/package_drazzy.com_index.json#L1765

```json
"url": "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2"
```

Has been changed to this:

```json
"url": "https://web.archive.org/web/20240701151914/https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2"
```

That would be great, except that these URLs download the Wayback Machine web page for the file instead of the file itself, causing the installation of ATTinyCore to fail with an error like this:

```text
Installing ATTinyCore:micronucleus@2.5-azd1b
Failed to install platform: 'ATTinyCore:avr:1.5.2'.
Error: 13 INTERNAL: Cannot install tool ATTinyCore:micronucleus@2.5-azd1b: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 12940 != 431381
```

(because the `ATTinyCore:micronucleus@2.5-azd1b` download URL pointed to a 12940 byte web page instead of the 431381 byte `micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2` file)

It is possible to avoid the modification of URLs in the file by using [this feature of Wayback Machine](https://en.wikipedia.org/wiki/Help:Using_the_Wayback_Machine#Specific_archive_copy:~:text=id_%20Identity%20%2D%20perform%20no%20alterations%20of%20the%20original%20resource%2C%20return%20it%20as%20it%20was%20archived.):

> `id_` Identity - perform no alterations of the original resource, return it as it was archived.

so the URL would be:

https://web.archive.org/web/20240701151914id_/https://drazzy.com/package_drazzy.com_index.json

(note the added `id_`)

But now the installation of ATTinyCore just fails with a different error:

```text
ATTinyCore:micronucleus@2.5-azd1b
Failed to install platform: 'ATTinyCore:avr:1.5.2'.
Error: 2 UNKNOWN: Get "https://azduino.com/bin/micronucleus/micronucleus-cli-2.5-azd1-x86_64-linux-gnu.tar.bz2": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: current time 2024-11-13T05:06:35-08:00 is after 2024-10-22T18:59:06Z
```

The reason is that now the URLs in the file point to the original locations and this file is hosted on azduino.com, which has the same problem of an expired certificate as drazzy.com. And anyway, if all we needed was another source for the hosted JSON file alone we could just use the "raw" URL for the [source file](https://github.com/SpenceKonde/ReleaseScripts/blob/master/package_drazzy.com_index.json) in the [`SpenceKonde/ReleaseScripts`](https://github.com/SpenceKonde/ReleaseScripts) GitHub repository:

https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/refs/heads/master/package_drazzy.com_index.json

kiwi64ajs, 2024-11-13:
> @kiwi64ajs did you test it? That won't work for multiple reasons:

The test I did was to put that Web Archive URL for the JSON into my Arduino IDE, then I restarted it, and it didn't complain.

However, because I was already on the latest version, it didn't offer to upgrade, which didn't surprise me.

What I didn't do was force a downgrade to a previous version and then upgrade to the latest.

I mistakenly thought that solution might have been a low-effort workaround until you could resolve the core SSL issue on https://drazzy.com/.

> And anyway, if all we needed was another source for the hosted JSON file alone we could just use the "raw" URL for the [source [...]

Yes, awesome, that works - why didn't I think of that... ;)

I can downgrade and upgrade that package and it all seems to work fine - thanks

Regards
Alex Shepherd

Chewhacker, 2024-11-14:
Hello, hoping for a pointer as I'm new to Arduino...

I used https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/refs/heads/master/package_drazzy.com_index.json and then have gone to board manager to install it, and get an error. 
[image]
Is this because I need to select a lower version than 1.5.2? I can't work it out :-( Any advice appreciated.

fsievers, 2024-11-14:
You can also Download the JSON file to your local machine and change the URL in the preference to something like `file:///home/USERNAME/foo/bar/package_drazzy.com_index.json`. The drawback is that you have to manually update the file if it change. But it would work around any SSL verifcation problems.

kiwi64ajs, 2024-11-14:
> Hello, hoping for a pointer as I'm new to Arduino...
> 
> I used https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/refs/heads/master/package_drazzy.com_index.json

> Is this because I need to select a lower version than 1.5.2? I can't work it out :-( Any advice appreciated.
That is a local install error of some kind.

What Operating System and Version are you running? What version of Arduino IDE are you using?

nomakewan, 2024-11-17:
> Hello, hoping for a pointer as I'm new to Arduino...
>
> I used https://raw.githubusercontent.com/SpenceKonde/ReleaseScripts/refs/heads/master/package_drazzy.com_index.json and then have gone to board [...]

This is happening to you because the certificate for `azduino.com` has expired, and the micronucleus toolchain is hosted on that website. Fixing the expired JSON won't help you.

per1234, 2024-11-22:
I don't want to do it here because I don't think issue reports are an appropriate place for providing technical support to users, but if anyone wants help installing ATTinyCore you are welcome to post over on Arduino Forum:

https://forum.arduino.cc/c/software/ide-2-x/93

jmerc77, 2024-11-27:
problem still exists. however, its expired. not missing. it seems like there's a misunderstanding. if it is missing, there would be a different error, I believe. if its expired like the error states, it needs renewal. but better yet would be to accept that machines fail an having more alternative options is always a good idea. even though you may not like putting the files here, that could be one of a few options. also, github is not the only git compatible repository site. there's one called git in fact. running a server solo is nearly impossible these days without using cloud compute. i do not expect anyone to do so without knowing what they're in for. doing so on the intra-net is different though.
simply put, "don't put your eggs in one basket."

jmerc77, 2024-11-28:
anyone thought of forking and adding a fix? may work as a different workaround that would be more secure.

jmerc77, 2024-12-04:
good news! it seems another user got the issue more recently and was solved shortly after. suggest trying to install again. should work now (i haven't but will in a few minutes. i'll report my findings.)

jmerc77, 2024-12-04:
yes it now works when installing. tested in arduino 2.3.4

kiwi64ajs, 2024-12-04:
I have confirmed that it also worked for me. I removed and reinstalled ATTinyCore, and it seemed to be normal

Alex

jvasileff, 2025-02-13:
It is, again, expired.

[image]

kiwi64ajs, 2025-02-13:
> It is, again, expired.
>
> Screenshot.2025-02-12.at.7.51.29.PM.png (view on web) <https://github.com/user-attachments/assets/94033cf3-e822-442b-a0e6-25b9b7e16aa0>

Try using: https://raw.githubusercontent.com/semir-t/attinycore/refs/heads/main/package_drazzy.com_index.json

per1234, 2025-02-17:
There is a dedicated report for the current occurrence of the problem here:

https://github.com/SpenceKonde/DxCore/issues/560

As I mentioned previously, until the certificate is renewed, we're always happy to help affected users to perform appropriate workaround for the problem over on Arduino Forum:

https://forum.arduino.cc/t/issue-accessing-http-drazzy-com-package-drazzy-com-index-json/1354916