SpenceKonde/ATTinyCore #886: There is no model ATtiny1604-SSN.

mayjack0312, 2024-10-31:
I installed it manually, but after the installation was successful, there was no development board model I needed. The model I needed was: ATtiny1604-SSN
Please tell me how should I operate?
[image]

pfoun, 2024-10-31:
I think you might need to install megaTinyCore.

closed 2024-11-01 by mayjack0312

SpenceKonde, 2024-11-01:
ATTinyCore is for Pre-AVRxt instruction set AVR (it's AVRe).  Last AVRe parts released, if I'm not mistaken were the m328PB or and m324PB? The latest tinyAVR classic parts (supported by ATTC) were the 441, 841, 828 and the 1634. 

All modern AVRs with part numbers starting with 1 or 2 digits which are a power of two (this is the flash size in kb), followed by a 0, 1, or 2 (indicating series - 0 = gimped 1-series, 1-series = quite capable, but has odd quirks, 2=series = how the Creator intended it), and finally a 2, 4, 6 or 7 for 8, 14, 20, or 24 pins - these are all modern tinyAVRs. 

All modern full size parts (except the early megas, ex uno wifi rev 2 `*` with the 4809 that launched as microchip was still trying to get Atmel down it's gullet. the modern ATmegas - nobody likes them too much. If you want a big un' you get a Dx, if you want it cheap, why get a 32-48 pin chip, yaknow? 

`*` BTW, the official designs for the Mega4809 Arduino.... I dunno... like, they had a list of mistakes to be careful to avoid, but they missed the word avoid or something.