SpenceKonde/ATTinyCore #891: Problem Installing ATTinyCore on Arduino IDE 2.3.3

VitorOM, 2024-11-29:
Hello,

I am encountering an issue while trying to install the ATTinyCore platform on the latest version of the Arduino IDE (2.3.3). When I click "Install" in the Board Manager, I receive the following error message:

[image]
[image]

SpenceKonde, 2024-12-03:
fixed, had forgotten to restart the second server after renewing cert for
it and rebooting it to make it pick up the recent kernal updates I ran to
keep it up to date (I've seen what happens to linux systems that don't get
this treatment - evetually the cert they have to verify SSL against
expires, andf you con't uprade them because it can't make an SSL connection
because it thinks the SSL server is compromised, and you need to either get
wizards involved to correct it, or you blow it away and rebuild.  It took
me over a month to do that last summer when i accidentally deleted the
running apache binary out from under it which kept working until the cert
ran out). I lost my main development system while still finishing up a
complicated tax issue that's occupied the last 8 months of most ob my
coherent thought, coming on the heels of a multiphase sequence of "random"
occurrences all of which were maximally disruptive to my specific modes of
thought and being.

One of those included loss (failure, not theft) of my main computer, with
all the infrastructure to do maintenance on my cloud servers set up on it,
leading to along search for the certificate on a backup so tht I could log
in, because I had also lost the data that I needed to upload if I had to
build th server, the only option if you lose your cert. (that was the
earlier outage in the fall). I would have just restored that from my file
server, except that the NAS box was shutting down before it could finish
boot due to whart turned out to be a defective connector terminal, fixed
last week, and problems in the network have now been mostly resolves with
the possible exception of the 2.5gbit ethernet adapter at this endpoint.

Hopefully I should be able to log into github again later today, cause the
phone is now starting the bootup process, finally. Then, I can start fixing
things in the core itself (at present, this is all I can do, I can't commit
changes oruse the web interface), otherwise,well, I'd need a new phone to
receive the 2FA

And I still haven't paid for my three year old phone completely, using
their default no-added-free. It crapped out a year ago, and when it is made
to turn on (hard to get it to not boot into recovery sometimes) there's no
sound, and if you try to cange the volume, it immediately goes to zero
again because it thinks the volume down button is continually pressed. I am
so ripshit over this that I vowed to cut my nose off to spite my face. I
refuse to buy another one of those fucking things until the one I'm paying
for has been paid off normally. It used to be these phones came with the
plan, They crapped out in 2 years, and that was okay because you got
another with the plan renewal every 2 years. Now, barring real estate and
automotive, they're among the most expensive items we own. Like, it's
number 2? Oscilloscope (4 channel) is number 1, the computer is cheaper
than the phone (and at least my computer was nearing end of usable
life anyway due to failing peripherals- it hadn't had working keyboard or
touchpad for 2 years, after the third keyboard replacement, the shop said
that they had to fight to get that last replacement, because they're out of
production. Two months later that keyboard failed too, so it had a keyboard
stuck to it, no battery life. Was on it's second SSD, after I had to pay
$600 to get all of my pcvb design files recovered from the old SSD  two
years ago. SSD still works, that one's going in the file server to act as
write cache now that I got the 2.5 gbit network between my laptop and the
mirrored raid array that I store my buik data on almost working (I think
the 2.5gbe network adapter is malfunctioning and dropping connection
intermittently, but the speed of data transfers when it works is epic. It's
good with 1GBit, but still network bound, with 2.5gbit it should run at the
speed of the drives.

Due to the frequency of ssd failures on laptops (the laptop before that one
had had 2 motherboard failures, each of which happened to the same chip on
the motherboard, the SSD interface, causing the data to be lost and the
SSDs to be unusable. So I deciuded to put most important data onto a file
server. After having to replace an SSD 6 times in the past 1.5 hardware
cycles, can you blame me for this revelation?  Or the poetic declaration:

"That which is of value shalt never reside upon a laptop SSD alone, for
they are weak and of the Devil. We shell prostrate ourselves before G--gle
and by His grace back up our documents in G--gle's Holy Cloud as they are
created! We already make monthly sacrifices to G--hub, and ever shall we,
[... 19 lines cut ...]
daren't even allude to have shown the path forward, and that it is revealed
within The Sixth Book of RAID...." (end prophetic rambling)

Where was I.... Cellphone prices, yes.... Now, you have to pay for it.
People balked at the prices per month for 2 year replacement, (because
phone makers have gone berko with cost and features which nobody wants and
people don't knotice except by the fact that their new phone seems no
faster than their old one. As always, what process tech giveth, programer's
taketh away.), so the payment is over like 3-4 years. The phones are still
designed to last at most 2, and most struggle to do that in real world
conditions. The forces of evil grow stronger every day. Doom approaches.
Pray for deliverance! Pray to G--gle! To G--hub! To L--ux! To G-U! To A--L
and T--C and I--el! Lest we all be doomed!
[... cut ...]
one....)

Anyway with that and a few more steps, I should be able to upload to
devices again and  return to being Spence instead of a ghost.
*No,you should be able to upload to one device at a time again, all
your USB ports are full except the stupid C type ones, remember? both your
other USB hubs are in pieces waiting for replacement parts to fix them*
Er, right, what he said, whatever said that. The parts have shipped
though... Aliexpress, should be here any day!

-Spence - sorry for the rant.

hmeijdam, 2024-12-03:
Did you buy a nice new keyboard for yourself and was this the test-drive? I am used to see waaaaay more typo's when you rant.

jmerc77, 2024-12-04:
wait... is it working now? not sure if me or others got the message.

jmerc77, 2024-12-04:
great! looks like it does install now. i just tried it (on Arduino 2.3.4 that recently was released). next steps for the owner is to close relating ssl issues. however, the drivers themselves are not signed by Microsoft and may lead to concerns form some (i'm on windows).

terrypin999, 2025-01-18:
Hobbyist, not programmer here. Could someone spell out the current status please? Is all the licence stuff sorted? Should I reinstall to get the very latest fixes, at least for Tiny85 and 88? Is this true for either my preferred  IDE v 1.8.19 and v 2?

Using help from the Arduino Forum experts I’ve successfully programmed my Tiny85 chips and Tiny88 boards, using Uni as ISO, but after many hour still unable to get any of my Tiny88 28/pin chips programmed.

hmeijdam, 2025-01-18:
If your current setup can program a ATT85, your setup can program a bare ATT88 chip as well as it works the same.

Are you using these settings to do a "burn bootloader"?

[image]

terrypin999, 2025-01-18:
Thanks. I've used that and the Optiboot option without success. Just repeated with those exact (default) settings.  using Burn Bootloader gives the errror summary
```
avrdude: Device signature = 0x1e930a
avrdude: Expected signature for ATtiny88 is 1E 93 11
```
Full error report below. It's the same as I get with Upload Using Pogrammer.

```
Arduino: 1.8.19 (Windows 10), Board: "ATtiny48/88 (No bootloader), Enabled, ATtiny88, 8 MHz (internal), Standard, EEPROM retained, B.O.D. Disabled (saves power), Enabled, Port A (pins 23~26)"

C:\Users\terry\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\terry\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf -v -pattiny88 -cstk500v1 -PCOM7 -b19200 -e -Uefuse:w:0xff:m -Uhfuse:w:0b11010111:m -Ulfuse:w:0xE2:m -Uflash:w:C:\Users\terry\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/bootloaders/empty/empty_all.hex:i 

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\terry\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.5.2/avrdude.conf"

         Using Port                    : COM7

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         Setting bit clk period        : 5.0

         AVR Part                      : ATtiny88

         Chip Erase delay              : 15000 us

         PAGEL                         : PD7

         BS2                           : PC2

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff

           flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Error while burning bootloader.

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e930a

avrdude: Expected signature for ATtiny88 is 1E 93 11

         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
```

hmeijdam, 2025-01-18:
> (avrdude: Device signature = 0x1e930a)

That is the device signature of a [Atmega88](https://onlinedocs.microchip.com/oxy/GUID-EDDC5F14-E7DB-4996-8AE5-E37AE12FEF7F-en-US-2/GUID-9962F7AD-84C2-4AB8-ACA4-8AAF009AD788.html)

You don't seem to have a Attiny88. Can you post a closeup picture?

terrypin999, 2025-01-18:
Embarrasing! You're absolutely right. I bought them impulsively from AliExpress thinking they were aTTiny88 chips, 28-pin DILs. After succeeding with the Tiny85 I was looking forward to doing the same with these much more versatile devices ;-(

[image]

Thanks for solving the puzzle, on which I wasted a full day.

hmeijdam, 2025-01-18:
Just install [Minicore](https://github.com/MCUdude/MiniCore) and you are good to go with the Atmega88

ahmedibrrahim, 2025-10-24:
> Hello,
> 
> I am encountering an issue while trying to install the ATTinyCore platform on the latest version of the Arduino IDE (2.3.3). When I click "Install" in the Board Manager, I receive the following error message:
> 
> [image] [image]

I'm facing the same issues. And can't install ATTinyCore on Arduino IDE. I tried to install an old version (1.3.0) from the boards manager and it works fine. But, it does not support ATTiny85 Micronucleus/Digispark, which I use. Once I select any version above 1.3.0, it raises the same error "Failed to install error unknown". Therefore, I am unable to install version 1.5.2, which is weird!

terrypin999, 2025-10-24:
> <https://github.com/user-attachments/assets/dd151411-df43-4570-afa1-850237ece5da> 2FA8B62A-F1FD-4001-96ED-B009B299AA6A.png

I’m puzzled why you have two versions of the same URL, one secure the
other not?