SpenceKonde/ATTinyCore #898: Third party  source  added to board manager but cant find attiny core library anymore

lupusworax, 2025-02-22:
seems to be a very recent problem. 
Worked flawless just a few days ago, now the attinycore library seems to be gone.  
adding the source seems like normal:

[image]
but when I search for attinycore in library manger nothing comes up: 

[image]

mikedevnull, 2025-02-23:
I also stumbled upon this.
I think the problem is that that the SSL certificate for `https://drazzy.com/` expired on February 12th. (It automatically redirects from http://drazzy.com to https://drazzy.com)

hmeijdam, 2025-02-23:
There is a [workaround](https://github.com/SpenceKonde/megaTinyCore/issues/1059#issuecomment-1950286469)

per1234, 2026-08-15:
The problem with the drazzy.com and/or azduino.com website used to host files downloaded by Arduino development tools while installing ATTinyCore is currently fixed. So it should now be possible to install ATTinyCore by following the standard procedure:

https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/Installation.md#boards-manager-installation