SpenceKonde/ATTinyCore #903: Blink sketch Attiny84a delay issue

ITstreet1, 2025-03-28:
I uploaded a simple Blink sketch on an Attiny84a using Arduino as ISP, and the upload went fine. No bootloader, just a sketch. I didn't touch anything from the options. The problem is, it doesn't blink every second, but on some 7-8 secs. Could you tell me what I did wrong?

hmeijdam, 2025-03-28:
You forgot to give the "Burn Bootloader" command, which will set the clock fuses to the 8MHz you selected. It is now running at 1MHz, which is the factory default speed after production.

And "yes" you need to do "burn bootloader" even without using a bootloader.

ITstreet1, 2025-03-29:
This did the trick. Can you shed some light on why I need a bootloader if I will not use it? The bootloader uses some amount of memory. 
What is the purpose of the bootloader on this IC if I still use SPI to program it? Couldn't the fuses be set just like on Atmega328p if I don't want the bootloader? Say, an option to set the clock before upload?

hmeijdam, 2025-03-29:
If you select Attiny84a (No Bootloader) 

[image]

and you press "Burn Bootloader" the IDE will just set the clock source and burn **no** bootloader.

When using a bootloader you can print and upload with the same serial adapter and it is using fewer pins. At the cost of giving up some memory and having a slower startup of your sketch.

On Atmega328p you also need to use the burn bootloader command (via SPI) to change it's clock, as the clock source change is fuse driven. 

Both Atmega328p as well Attiny84a make use of AVRDUDE as the programming software, so you could just provide the AVRDUDE command to set the clock fuses. However if you make a mistake with the fuse values you could brick the chip, making it impossible to further program it, without a High Voltage programmer.

ITstreet1, 2025-03-30:
By using serial adapter, you mean UART? I don't see any RX/TX pins on this chip.

hmeijdam, 2025-03-30:
Did you look at the right side of the [picture of the pins](https://github.com/SpenceKonde/ATTinyCore/blob/v2.0.0-devThis-is-the-head-submit-PRs-against-this/avr/extras/ATtiny_x4.md)?

PA1 and PA2